platformio/platformio-atom-ide #1254: Uncaught Error: Could not start PIO Home server: Error: Could not install 'pysite-pioplus' with v...

thiaguim18, 2017-12-10:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.22.1 x64
**Electron**: 1.6.15
**OS**: Unknown Windows version
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 2.0.0-rc.4

### Stack Trace

Uncaught Error: Could not start PIO Home server: Error: Could not install 'pysite-pioplus' with version requirements '>=0.4.2,<2' for your system 'windows_amd64'.
If you use Antivirus, it can block PlatformIO Package Manager. Try to disable it for a while.

```
At C:\Users\Usuario\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1571

Error: Could not start PIO Home server: Error: Could not install 'pysite-pioplus' with version requirements '>=0.4.2,<2' for your system 'windows_amd64'.
If you use Antivirus, it can block PlatformIO Package Manager. Try to disable it for a while.

    at /packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1571:23
    at ChildProcess.onExit (/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:499:5)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at ChildProcess.cp.emit (/packages/platformio-ide/node_modules/cross-spawn/lib/enoent.js:40:29)
    at maybeClose (internal/child_process.js:885:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
```

### Commands

```
     -2:49.8.0 tree-view:show (atom-workspace.workspace.scrollbars-visible-always)
     -2:06.2.0 platformio-ide:target:build (atom-pane.pane.active)
  3x -1:49.4.0 application:show-settings (atom-pane.pane.active)
  2x -1:43 platformio-ide:maintenance.serial-monitor (div.package-detail.panels-item)
  2x -1:06.2.0 project-find:show (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
  2x -1:03.1.0 build:toggle-panel (input.hidden-input)
     -0:58.9.0 project-find:show (input.hidden-input)
     -0:47.7.0 platformio-ide:settings:pkg-platformio-ide (div.package-detail.panels-item)
     -0:31.7.0 platformio-ide:home (div.package-detail.panels-item)
```

### Non-Core Packages

```
autocomplete-clang 0.11.4 
build 0.69.0 
busy-signal 1.4.3 
file-icons 2.1.14 
intentions 1.1.5 
language-ini 1.19.0 
linter 2.2.0 
linter-gcc 0.7.1 
linter-ui-default 1.6.10 
platformio-ide 2.0.0-rc.4 
platformio-ide-debugger 1.2.4 
platformio-ide-terminal 2.7.0 
tool-bar 1.1.0 
```

ivankravets, 2017-12-15:
Please run in PIO IDE Terminal `pio home`. Does it work now?

ivankravets, 2017-12-15:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/1227

closed 2017-12-15 by ivankravets