platformio/platformio-atom-ide #1255: InstallationManager: Just downloaded latest Atom and installed platformio-ide macOs Sierra

marinehero, 2017-12-11:
Conda Virtualenv: 
PackageNotFoundError: Packages missing in current channels:
            
  - python 2*

We have searched for the packages in the following channels:
            
  - https://repo.continuum.io/pkgs/pro/osx-64
  - https://repo.continuum.io/pkgs/pro/noarch
  - https://conda.anaconda.org/conda-canary/osx-64
  - https://conda.anaconda.org/conda-canary/noarch

ivankravets, 2017-12-15:
Do you use the latest `conda` in a system? Did you install it?

marinehero, 2017-12-15:
If I remove the following from my path then VSCode/PlatformIO works

> # added by Anaconda 1.9.1 installer
> #export PATH="/Users/james/anaconda/bin:$PATH"

and

> # added by Anaconda2 5.0.1 installer
> #export PATH="/Users/james/anaconda2/bin:$PATH"

weird...

ivankravets, 2017-12-15:
Does it work now?

marinehero, 2017-12-15:
It works if I un-comment  the above 2. i.e With no conda installed.

uname -a

`Darwin Jamess-MacBook-Pro.local 17.3.0 Darwin Kernel Version 17.3.0: Thu Nov  9 18:09:22 PST 2017; root:xnu-4570.31.3~1/RELEASE_X86_64 x86_64`

closed 2018-06-25 by ivankravets