platformio/platformio-atom-ide #1257: InstallationManager

shi27feng, 2017-12-11:
PIP: Exception:
Traceback (most recent call last):
  File "/home/feng/.platformio/penv/local/lib/python2.7/site-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/home/feng/.platformio/penv/local/lib/python2.7/site-packages/pip/commands/install.py", line 272, in run
    with self._build_session(options) as session:
  File "/home/feng/.platformio/penv/local/lib/python2.7/site-packages/pip/basecommand.py", line 72, in _build_session
    insecure_hosts=options.trusted_hosts,
  File "/home/feng/.platformio/penv/local/lib/python2.7/site-packages/pip/download.py", line 329, in __init__
    self.headers["User-Agent"] = user_agent()
  File "/home/feng/.platformio/penv/local/lib/python2.7/site-packages/pip/download.py", line 93, in user_agent
    from pip._vendor import distro
  File "/home/feng/.platformio/penv/local/lib/python2.7/site-packages/pip/_vendor/distro.py", line 1050, in <module>
    _distro = LinuxDistribution()
  File "/home/feng/.platformio/penv/local/lib/python2.7/site-packages/pip/_vendor/distro.py", line 594, in __init__
    if include_lsb else {}
  File "/home/feng/.platformio/penv/local/lib/python2.7/site-packages/pip/_vendor/distro.py", line 933, in _get_lsb_release_info
    raise subprocess.CalledProcessError(code, cmd, stdout)
CalledProcessError: Command 'lsb_release -a' returned non-zero exit status 1
Traceback (most recent call last):
  File "/home/feng/.platformio/penv/bin/pip", line 11, in <module>
    sys.exit(main())
  File "/home/feng/.platformio/penv/local/lib/python2.7/site-packages/pip/__init__.py", line 233, in main
    return command.main(cmd_args)
  File "/home/feng/.platformio/penv/local/lib/python2.7/site-packages/pip/basecommand.py", line 251, in main
    timeout=min(5, options.timeout)) as session:
  File "/home/feng/.platformio/penv/local/lib/python2.7/site-packages/pip/basecommand.py", line 72, in _build_session
    insecure_hosts=options.trusted_hosts,
  File "/home/feng/.platformio/penv/local/lib/python2.7/site-packages/pip/download.py", line 329, in __init__
    self.headers["User-Agent"] = user_agent()
  File "/home/feng/.platformio/penv/local/lib/python2.7/site-packages/pip/download.py", line 93, in user_agent
    from pip._vendor import distro
  File "/home/feng/.platformio/penv/local/lib/python2.7/site-packages/pip/_vendor/distro.py", line 1050, in <module>
    _distro = LinuxDistribution()
  File "/home/feng/.platformio/penv/local/lib/python2.7/site-packages/pip/_vendor/distro.py", line 594, in __init__
    if include_lsb else {}
  File "/home/feng/.platformio/penv/local/lib/python2.7/site-packages/pip/_vendor/distro.py", line 933, in _get_lsb_release_info
    raise subprocess.CalledProcessError(code, cmd, stdout)
subprocess.CalledProcessError: Command 'lsb_release -a' returned non-zero exit status 1

george-vieira, 2017-12-12:
are you running debian based distro? ubuntu etc?
have you got lsb-release installed?

ivankravets, 2018-01-10:
Please reopen if you still need help

closed 2018-01-10 by ivankravets

tejastank, 2021-02-27:
@george-vieira did you upgraded python version recently ?  do you have multiple version of python working ?

does most recent python multi-version activities happen on system ?