platformio/platformio-atom-ide #1258: C/C++ project index rebuild failed

johoberm, 2017-12-12:
Error: Please install Git client from https://git-scm.com/downloads

Git is installed via initial Setup (Windows Atom)

ivankravets, 2017-12-15:
Do you see it in PIO IDE Terminal?

```
git --version
```

cosmocracy, 2017-12-23:
I get this error, too, after fresh Atom/PlatformIO install on Windows 10.

`git --version` output:
```
git version 2.15.1.windows.2
```

cosmocracy, 2017-12-23:
Screenshot, below, shows error during build (in terminal window) as well as popup error boxes from IDE:

[image]

cosmocracy, 2017-12-23:
(And I restarted Atom after Git install (including multiple reinstall attempts))

ivankravets, 2017-12-23:
Please check that `git` is in your system PATH. Maybe, OS reboot will help.

closed 2018-06-21 by ivankravets