platformio/platformio-atom-ide #1259: Uncaught Error: Could not start PIO Home server: Traceback (most recent call last):  File "main....

rites1978, 2017-12-12:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.22.1 x64
**Electron**: 1.6.15
**OS**: Mac OS X 10.13.1
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 2.0.0-rc.4

### Stack Trace

Uncaught Error: Could not start PIO Home server: Traceback (most recent call last):
  File "main.py", line 16, in init main
  File "cmd_debug.py", line 21, in init cmd_debug
ImportError: No module named twisted.internet

```
At /Users/Riteshlal/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1571

Error: Could not start PIO Home server: Traceback (most recent call last):
  File "main.py", line 16, in init main
  File "cmd_debug.py", line 21, in init cmd_debug
ImportError: No module named twisted.internet

    at /packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1571:23
    at ChildProcess.onExit (/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:499:5)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at maybeClose (internal/child_process.js:885:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
```

### Commands

```
     -0:20.4.0 tree-view:show (atom-workspace.workspace.scrollbars-visible-when-scrolling)
```

### Non-Core Packages

```
autocomplete-clang 0.11.3 
build 0.69.0 
busy 0.7.0 
busy-signal 1.4.3 
file-icons 2.1.13 
intentions 1.1.5 
language-ini 1.19.0 
linter 2.2.0 
linter-gcc 0.7.1 
linter-ui-default 1.6.10 
platformio-ide 2.0.0-rc.4 
platformio-ide-debugger 1.2.4 
platformio-ide-terminal 2.7.0 
tool-bar 1.1.0 
```

gordonthree, 2017-12-15:
Me too. Fresh install of Atom and platformio under Windows 10 Pro x64 build 1709
Slightly newer Atom, 1.23.1 x64

ivankravets, 2017-12-15:
Please open PIO IDE Terminal and type

1. `pio update`
2. `pio home`

Does it work now?

gordonthree, 2017-12-15:
Hi Ivan,

pio update didn't find anything to update

pio home fails:

`PS C:\Users\Gordon> pio home
Traceback (most recent call last):
  File "main.py", line 16, in init main
  File "cmd_debug.py", line 21, in init cmd_debug
  File "C:\Users\Gordon\.platformio\packages\pysite-pioplus\twisted\__init__.py", line 23, in <module>
    from twisted._version import __version__ as version
  File "C:\Users\Gordon\.platformio\packages\pysite-pioplus\twisted\_version.py", line 8, in <module>
    from incremental import Version
  File "C:\Users\Gordon\.platformio\packages\pysite-pioplus\incremental\__init__.py", line 1
SyntaxError: Non-ASCII character '\xc0' in file C:\Users\Gordon\.platformio\packages\pysite-pioplus\incremental\__init__.p
y on line 1, but no encoding declared; see http://python.org/dev/peps/pep-0263/ for details`

ivankravets, 2017-12-15:
Try to remove this folder `C:\Users\Gordon\.platformio\packages\pysite-pioplus` and start again.

gordonthree, 2017-12-15:
OK deleted that folder and started atom. The atom tab trying to load the platformio home screen sits there forever with a little spinner going around and around.

from the console however
PS C:\Users\Gordon> pio home
PlatformIO Plus (https://pioplus.com) v0.12.1

  ___I_
 /\-_--\   PlatformIO Home
/  \_-__\
|[]| [] |  http://127.0.0.1:8008
|__|____|_______________________

The home server does launch, and the home page opens in my default web browser.

From there, going back to Atom, I can go into PlatformIO menu and choose PlatformIO Home and it will open in an Atom tab,

ivankravets, 2017-12-15:
@
Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

gordonthree, 2017-12-17:
Interestingly, after having the laptop off for two days, powering up Atom to respond to this thread, no complaints from the code... odd?

echo %PATH%
C:\Program Files (x86)\Intel\iCLS Client\;C:\Program Files\Intel\iCLS Client\;C:\ProgramData\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\PuTTY\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files\Git\cmd;C:\Program Files\LLVM\bin;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\Intel\Intel(R) Management Engine Components\IPT;C:\Users\Gordon\AppData\Local\Microsoft\WindowsApps;C:\Program Files\Microsoft VS Code\bin;C:\Program Files (x86)\Sophos\Sophos SSL VPN Client\bin;C:\Users\Gordon\AppData\Local\atom\bin

C:\Users\Gordon>echo %PYTHONPATH%
%PYTHONPATH%

C:\Users\Gordon>python --version
'python' is not recognized as an internal or external command, operable program or batch file.

Same deal for PIP.  I have not installed python on this machine. If it is present behind the scenes, it is being called directly by whatever program installed it.

ivankravets, 2018-02-17:
Please reopen if you still need help

closed 2018-02-17 by ivankravets

codebrainz, 2020-08-15:
I had this issue (except with VSCode). Fixed it by running `pip install twisted` from Windows command-line - Twisted isn't installed by default on any OS I know of. Unfortunately Twisted requires first installing Microsoft Build Tools which is massive/slow to install, but once that was all done, I deleted the `.platformio` folder in my home directory and started VSCode and once PlatformIO finished installing itself again and then prompted me to restart VSCode, doing so allowed it to work. I suspect it's the same issue for Atom.