platformio/platformio-atom-ide #1260: C/C++ project index rebuild failed

swilalaa, 2017-12-12:
I'm trying to install platformio-ide on Atom and get following issues:

**C/C++ project index rebuild failed**
Error: spawn platformio ENOENT

**InstallationManager**
PIP: Command "/Users/KS/.platformio/penv/bin/python -u -c "import setuptools, tokenize;__file__='/private/var/folders/74/717cmqhs4wb2sv408dzl7hkh0000gn/T/pip-KzOqq9-build/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /var/folders/74/717cmqhs4wb2sv408dzl7hkh0000gn/T/pip-YRRHes-record/install-record.txt --single-version-externally-managed --compile --install-headers /Users/KS/.platformio/penv/bin/../include/site/python2.7/platformio" failed with error code 1 in /private/var/folders/74/717cmqhs4wb2sv408dzl7hkh0000gn/T/pip-KzOqq9-build/

My Environment:

MacOS 10.12.6
phyton 2.7.14
pip (9.0.1)
setuptools (38.2.4)

Any Ideas what the issue may be?

Thanks in advance!

ivankravets, 2017-12-15:
Please provide an output from a system Terminal
```
echo $PATH
echo $PYTHONPATH
python --version
pip --version
pip list
```

swilalaa, 2017-12-15:
KS-Air:~ KS$ echo $PATH
/Library/Frameworks/Python.framework/Versions/2.7/bin:/Library/Frameworks/Python.framework/Versions/2.7/bin:/Library/Frameworks/Python.framework/Versions/2.7/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin

KS-Air:~ KS$ echo $PYTHONPATH
(Empty)

KS-Air:~ KS$ python --version
Python 2.7.14

KS-Air:~ KS$ pip --version
pip 9.0.1 from /Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages (python 2.7)

KS-Air:~ KS$ pip list
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
pip (9.0.1)
setuptools (38.2.4)

ivankravets, 2017-12-15:
Could you remove `/Users/KS/.platformio/penv` and restart IDE? Does it work now?

swilalaa, 2017-12-15:
this solved the issue, perfect thank you @ivankravets ! can you help me understand what caused the issue?

closed 2017-12-15 by swilalaa

ivankravets, 2017-12-15:
Did you update Python to 2.7.14 after PIO installation?