platformio/platformio-atom-ide #1261: C/C++ project index rebuild failed

weberseb, 2017-12-12:
Error: Please manually remove file `/Users/sebastianweber/Desktop/Sonoff-Tasmota-development/.pioenvs`
Can not remove temporary directory `/Users/sebastianweber/Desktop/Sonoff-Tasmota-development/.pioenvs`. Please remove `.pioenvs` directory from the project manually to avoid build issues

[Tue Dec 12 22:04:16 2017] Processing sonoff-DE (build_flags: -Wl,-Tesp8266.flash.1m0.ld -DMQTT_MAX_PACKET_SIZE=512 -DMY_LANGUAGE=de-DE; lib_deps: PubSubClient, NeoPixelBus, IRremoteESP8266, ArduinoJSON; platform: espressif8266; board: esp01_1m; framework: arduino; monitor_baud: 115200; board_flash_mode: dout)
--------------------------------------------------------------------------------
Exception: [Errno 13] Permission denied: '/Users/sebastianweber/Desktop/Sonoff-Tasmota-development/.pioenvs/do-not-modify-files-here.url':
File "/Users/sebastianweber/.platformio/penv/lib/python2.7/site-packages/platformio/builder/main.py", line 93:
PROJECTPIOENVS_DIR=util.get_projectpioenvs_dir(),
File "/Users/sebastianweber/.platformio/penv/lib/python2.7/site-packages/platformio/util.py", line 323:
raise Exception(e)
========================== [ERROR] Took 1.24 seconds ==========================

================================== [SUMMARY] ==================================
Environment sonoff        	[SKIP]
Environment sonoff-NL     	[SKIP]
Environment sonoff-DE     	[ERROR]
Environment sonoff-PL     	[SKIP]
Environment sonoff-minimal	[SKIP]
Environment sonoff-ds18x20	[SKIP]
========================== [ERROR] Took 1.24 seconds ==========================

ivankravets, 2017-12-12:
Can not remove temporary directory /Users/sebastianweber/Desktop/Sonoff-Tasmota-development/.pioenvs. Please remove .pioenvs directory from the project manually to avoid build issues

closed 2017-12-12 by ivankravets

jeylites, 2018-05-09:
I'm in the same situation. What does it mean?