platformio/platformio-atom-ide #1263: InstallationManager

mdanetzky, 2017-12-13:
Virtualenv Create: Traceback (most recent call last):
  File "/Users/matthias/.platformio/.cache/tmp-3372ozjoM2Y3D7wC/virtualenv-14.0.6/virtualenv.py", line 2304, in <module>
    main()
  File "/Users/matthias/.platformio/.cache/tmp-3372ozjoM2Y3D7wC/virtualenv-14.0.6/virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "/Users/matthias/.platformio/.cache/tmp-3372ozjoM2Y3D7wC/virtualenv-14.0.6/virtualenv.py", line 904, in create_environment
    site_packages=site_packages, clear=clear, symlink=symlink))
  File "/Users/matthias/.platformio/penv/lib/python2.7/posixpath.py", line 360, in abspath
    if not isabs(path):
  File "/Users/matthias/.platformio/penv/lib/python2.7/posixpath.py", line 54, in isabs
    return s.startswith('/')
AttributeError: 'NoneType' object has no attribute 'startswith'

ivankravets, 2017-12-14:
Please provide an output from a system Terminal
```
echo $PATH
echo $PYTHONPATH
python --version
pip --version
pip list
```

mdanetzky, 2017-12-14:
echo $PATH
/usr/local/opt/libxml2/bin:/usr/local/opt/python/libexec/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Users/matthias/n/bin

echo $PYTHONPATH

python --version
Python 2.7.10

pip --version
-bash: pip: command not found

matthias$ pip list
-bash: pip: command not found

ivankravets, 2017-12-14:
> /usr/local/opt/python

Do you have a custom Python in a system?

mdanetzky, 2017-12-14:
I just reinstalled OSX Sierra after botched update of High Sierra

ivankravets, 2017-12-14:
Could you provide TeamViewer session to debug this issue? me@ikravets.com

ivankravets, 2017-12-15:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/1240

closed 2017-12-15 by ivankravets