platformio/platformio-atom-ide #1265: C/C++ project index rebuild failed

CaptainRon47, 2017-12-14:
Error: Could not load broken JSON: C:\Users\Ron\GitHub\kbox-firmware-rz\.piolibdeps\_tmp_installing-lb5xs_-package\library.json

CaptainRon47, 2017-12-14:
performin a PI build this error showed up in PlatformIO home page?

ivankravets, 2017-12-15:
Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

CaptainRon47, 2017-12-15:
Ivan,

Below (also attached) is the output you requested. 

 

I have a question about the request to try the staging version.

Could you try staging version? http://docs.platformio.org/en/latest/platforms/espressif8266.html#using-arduino-framework-with-staging-version

 

The ‘esp’ environment of the KBox project program BUILDS without errors, it is the UPLOAD to the Teensy where I get the error.   The instructions you sited only compare the BUILD of the project under the two IDE’s (PlatformIO and Arduino), do you want me to try the UPLOAD in both as well?

 

I did modify the platformio.ini file for the [env:esp] section of the project as below (also attached) and that BUILT successfully, but it did so under the original platform statement (teensy) as well.

 

Thanks,

Ron

 

…

[env:esp]

src_filter = +<common/*>,+<esp/*>

# TODO: enable -Werror once platformio published v2.4 of esp8266/Arduino (pr #3271)

build_flags = -Wall -Isrc/common -Isrc/esp -DHAVE_STRLCPY -DHAVE_STRLCAT

platform = https://github.com/platformio/platform-espressif8266.git#feature/stage

framework = arduino

board = esp_wroom_02

extra_scripts = tools/platformio_cfg_esp.py

lib_ldf_mode = deep

lib_ignore = NMEA2000-Teensy, FlexCAN_Library, Time

lib_deps =

  ${common.lib_deps_common}

  Adafruit NeoPixel

  elapsedMillis

  https://github.com/me-no-dev/ESPAsyncTCP#a57560d

  https://github.com/me-no-dev/ESPAsyncWebServer#63b5303

…

 

 

Please provide an output from system Terminal cmd

echo %PATH%

[... 153 lines cut ...]
python-dateutil (2.6.1)
requests (2.18.4)
semantic-version (2.6.0)
setuptools (36.4.0)
six (1.11.0)
urllib3 (1.22)

C:\Users\Ron>

[env:esp]
src_filter = +<common/*>,+<esp/*>
# TODO: enable -Werror once platformio published v2.4 of esp8266/Arduino (pr #3271)
build_flags = -Wall -Isrc/common -Isrc/esp -DHAVE_STRLCPY -DHAVE_STRLCAT
platform = https://github.com/platformio/platform-espressif8266.git#feature/stage
framework = arduino
board = esp_wroom_02
extra_scripts = tools/platformio_cfg_esp.py
lib_ldf_mode = deep
lib_ignore = NMEA2000-Teensy, FlexCAN_Library, Time
lib_deps =
  ${common.lib_deps_common}
  Adafruit NeoPixel
  elapsedMillis
  https://github.com/me-no-dev/ESPAsyncTCP#a57560d
  https://github.com/me-no-dev/ESPAsyncWebServer#63b5303

CaptainRon47, 2017-12-15:
Ivan,
Sorry I answered both of your emails here, I should have realized they were for different issues.
Ron

ivankravets, 2017-12-15:
I don't understand your problem. Please provide a simple project to reproduce this issue.

ivankravets, 2017-12-15:
Please post 1 report to https://community.platformio.org

closed 2017-12-15 by ivankravets