platformio/platformio-atom-ide #1267: Uncaught Error: Could not start PIO Home server: ~/.platformio/packages/tool-pioplus/pioplus: sym...

s0170071, 2017-12-14:
[Enter steps to reproduce:]

1. fresh install of linux mint, then atom, then platformio-ide. Restart as prompted then the error.
2. ...

**Atom**: 1.23.1 x64
**Electron**: 1.6.15
**OS**: linux 4.10.0-38-generic
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 2.0.0-rc.4

### Stack Trace

Uncaught Error: Could not start PIO Home server: /home/john/.platformio/packages/tool-pioplus/pioplus: symbol lookup error: /home/john/.platformio/packages/tool-pioplus/pioplus: undefined symbol: PyUnicodeUCS4_FromStringAndSize

```
At /home/john/.var/app/io.atom.Atom/data/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1571

Error: Could not start PIO Home server: /home/john/.platformio/packages/tool-pioplus/pioplus: symbol lookup error: /home/john/.platformio/packages/tool-pioplus/pioplus: undefined symbol: PyUnicodeUCS4_FromStringAndSize

    at /packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1571:23
    at ChildProcess.onExit (/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:499:5)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at maybeClose (internal/child_process.js:885:16)
    at Socket.<anonymous> (internal/child_process.js:334:11)
    at emitOne (events.js:96:13)
    at Socket.emit (events.js:188:7)
    at Pipe._handle.close [as _onclose] (net.js:501:12)
```

### Commands

```
```

### Non-Core Packages

```
autocomplete-clang 0.11.4 
build 0.69.0 
busy-signal 1.4.3 
file-icons 2.1.14 
intentions 1.1.5 
language-ini 1.19.0 
linter 2.2.0 
linter-clang 4.1.2 
linter-gcc 0.7.1 
linter-ui-default 1.6.10 
platformio-ide 2.0.0-rc.4 
platformio-ide-debugger 1.2.4 
platformio-ide-terminal 2.7.0 
tool-bar 1.1.0 
```

ivankravets, 2017-12-15:
What is a type of your distro?

Please provide an output from a system Terminal
```
echo $PATH
echo $PYTHONPATH
python --version
pip --version
pip list
```

s0170071, 2017-12-15:
I use a fresh linux mint cinnamon. Installed just yesterday.

Here the output.:

john@Mint ~ $ echo $PATH
/home/john/bin:/home/john/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games
john@Mint ~ $ echo $PYTHONPATH

john@Mint ~ $ python --version
Python 2.7.12
john@Mint ~ $ pip --version
pip 9.0.1 from /home/john/.local/lib/python2.7/site-packages (python 2.7)
john@Mint ~ $ pip list
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
attrs (15.2.0)
BeautifulSoup (3.2.1)
blinker (1.3)
chardet (2.3.0)
configglue (1.1.2)
configobj (5.0.6)
cryptography (1.2.3)
debtags (2.0)
defer (1.0.6)
dirspec (13.10)
duplicity (0.7.6)
ecdsa (0.13)
enum34 (1.1.2)
feedparser (5.1.3)
httplib2 (0.9.1)
idna (2.0)
ipaddress (1.0.16)
lockfile (0.12.2)
lxml (3.5.0)
Mako (1.0.3)
MarkupSafe (0.23)
nemo-emblems (3.6.0)
netifaces (0.10.4)
oauthlib (1.0.3)
oneconf (0.3.9)
PAM (0.4.2)
paramiko (1.16.0)
pexpect (4.0.1)
Pillow (3.1.2)
pip (9.0.1)
piston-mini-client (0.7.5)
protobuf (2.6.1)
ptyprocess (0.5)
pyasn1 (0.1.9)
pyasn1-modules (0.0.7)
pycrypto (2.6.1)
pycups (1.9.73)
pycurl (7.43.0)
pygobject (3.20.0)
pyinotify (0.9.6)
PyJWT (1.3.0)
pyOpenSSL (0.15.1)
pyparted (3.10.7)
pyserial (3.0.1)
pysmbc (1.0.15.5)
python-apt (1.1.0b1)
python-debian (0.1.27)
python-xlib (0.14)
pyxdg (0.25)
reportlab (3.3.0)
service-identity (16.0.0)
setproctitle (1.1.8)
six (1.10.0)
Twisted (16.0.0)
uTidylib (0.2)
zope.interface (4.1.3)

majekw, 2017-12-20:
Same here on Slackware64 14.2:

**Atom**: 1.23.1 x64
**Electron**: 1.6.15
**OS**: linux 4.9.58
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 2.0.0-rc.4

### Stack Trace

Uncaught Error: Could not start PIO Home server: /home/majek/.platformio/packages/tool-pioplus/pioplus: symbol lookup error: /home/majek/.platformio/packages/tool-pioplus/pioplus: undefined symbol: PyUnicodeUCS4_FromStringAndSize

```
At /home/majek/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1571

Error: Could not start PIO Home server: /home/majek/.platformio/packages/tool-pioplus/pioplus: symbol lookup error: /home/majek/.platformio/packages/tool-pioplus/pioplus: undefined symbol: PyUnicodeUCS4_FromStringAndSize

    at /packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1571:23
    at ChildProcess.onExit (/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:499:5)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at maybeClose (internal/child_process.js:885:16)
    at Socket.<anonymous> (internal/child_process.js:334:11)
    at emitOne (events.js:96:13)
    at Socket.emit (events.js:188:7)
    at Pipe._handle.close [as _onclose] (net.js:501:12)
```

### Commands

```
```

### Non-Core Packages

```
autocomplete-clang 0.11.4 
build 0.70.0 
busy-signal 1.4.3 
file-icons 2.1.15 
intentions 1.1.5 
language-ini 1.19.0 
linter 2.2.0 
linter-gcc 0.7.1 
linter-ui-default 1.6.10 
platformio-ide 2.0.0-rc.4 
platformio-ide-debugger 1.2.4 
platformio-ide-terminal 2.7.0 
tool-bar 1.1.0 
```
echo $PATH
/usr/lib64/go1.8.5/go/bin:/usr/local/bin:/usr/bin:/bin:/usr/games:/usr/lib64/java/bin:/usr/lib64/java/jre/bin:/usr/lib64/kde4/libexec:/usr/lib64/qt/bin:/usr/lib64/qt5/bin:/usr/share/texmf/bin:/home/majek/node_modules/.bin:/home/majek/.local/bin
majek@lechtacz:~$ echo $PYTHONPATH

majek@lechtacz:~$ python --version
Python 2.7.14
majek@lechtacz:~$ pip --version
pip 9.0.1 from /usr/lib64/python2.7/site-packages/pip-9.0.1-py2.7.egg (python 2.7)
majek@lechtacz:~$ pip list
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.                                                                                                                            
ansible (2.4.2.0)
asn1crypto (0.21.1)
awscli (1.11.100)
botocore (1.5.63)
cffi (1.11.2)
colorama (0.3.7)
cryptography (1.8.1)
cupshelpers (1.0)
Cython (0.23.4)
docutils (0.13.1)
enum34 (1.1.6)
futures (3.1.1)
getmail (4.54.0)
httplib2 (0.10.3)
idna (2.5)
include-server (3.1)
ipaddress (1.0.18)
isc (1.0)
Jinja2 (2.9.5)
jmespath (0.9.3)
libsigrok (0.6.0)
libwebp (0.0)
M2Crypto (0.23.0)
MarkupSafe (1.0)
mercurial (4.3.1)
netsnmp-python (1.0a1)
numpy (1.11.1)
packaging (16.8)
paramiko (2.1.2)
Pillow (3.0.0)
pip (9.0.1)
pssh (2.3.1)
pyasn1 (0.2.3)
pycparser (2.18)
pycups (1.9.73)
pycurl (7.43.0)
pygobject (3.18.2)
pyparsing (2.1.8)
Pyrex (0.9.9)
python-dateutil (2.6.0)
python-libpisock (0.12.5)
python-sane (2.8.2)
PyYAML (3.12)
rsa (3.4.2)
s3transfer (0.1.10)
scons (2.4.1)
setuptools (22.0.5)
six (1.11.0)
urwid (1.0.3)
vboxapi (1.0)
wicd (1.7.4)
zenmap (7.12)

I have the same issue also on Atom 1.22.1.

s0170071, 2018-01-10:
The new version of Atom 1.23.3 fixed that issue for me.
Just install the new version (then you have both on your system) and then delete the 1.23.1

ivankravets, 2018-01-10:
@s0170071 did you update system's python? The problem is with Python, not Atom

s0170071, 2018-01-10:
not tham I am aware of. What can I do to check ?

ivankravets, 2018-01-10:
@majekw have you resolved this issue? Could you also upgrade your Atom editor?

closed 2018-06-21 by ivankravets