platformio/platformio-atom-ide #1271: C/C++ project index rebuild failed

thetonto, 2017-12-16:
Cloning into 'C:\Users\the_d\Documents\PlatformIO\Projects\171216-211454-nodemcuv2\.piolibdeps\_tmp_installing-5wnwtw-package'...
Note: checking out '76e00a046255ae4ac9f61f36af90d8c5fc9828c9'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

Error: Traceback (most recent call last):
  File "c:\users\the_d\.platformio\penv\lib\site-packages\platformio\__main__.py", line 120, in main
    cli(None, None, None)
  File "c:\users\the_d\.platformio\penv\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\the_d\.platformio\penv\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\users\the_d\.platformio\penv\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\the_d\.platformio\penv\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\the_d\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\users\the_d\.platformio\penv\lib\site-packages\click\decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\the_d\.platformio\penv\lib\site-packages\platformio\commands\init.py", line 102, in cli
    pg = ProjectGenerator(project_dir, ide, env_name)
  File "c:\users\the_d\.platformio\penv\lib\site-packages\platformio\ide\projectgenerator.py", line 38, in __init__
    self._gather_tplvars()
  File "c:\users\the_d\.platformio\penv\lib\site-packages\platformio\ide\projectgenerator.py", line 155, in _gather_tplvars
    self._tplvars.update(self.get_project_build_data())
  File "c:\users\the_d\.platformio\penv\lib\site-packages\platformio\util.py", line 134, in __call__
    value = self.func(*args)
  File "c:\users\the_d\.platformio\penv\lib\site-packages\platformio\ide\projectgenerator.py", line 79, in get_project_build_data
    raise result.exception
WindowsError: [Error 183] Cannot create a file when that file already exists

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

ivankravets, 2017-12-16:
Could you share your `platformio.ini`?

thetonto, 2017-12-16:
Morning.  
Ini file as requested:

[env:nodemcuv2]
platform = https://github.com/platformio/platform-espressif8266.git#feature/stage
board = nodemcuv2
framework = arduino

lib_deps = https://github.com/marvinroger/homie-esp8266.git#v2.0.0-beta.2

ivankravets, 2017-12-16:
Could you reproduce this issue again?

thetonto, 2017-12-16:
Tried, but I get inconsistent results from each build.  Latest is this.  I have unintalled ATOM removed all the directorys and re-installed.  Still errors.  Switched back to using Arduino as that is very solid, thou would like to understand why this does not work. 

Following this guide:
http://marvinroger.github.io/homie-esp8266/docs/2.0.0-beta.2/quickstart/getting-started/

Compiling .pioenvs\nodemcuv2\lib\ESPAsyncTCP_ID305\ESPAsyncTCP.o
Compiling .pioenvs\nodemcuv2\lib\ESPAsyncTCP_ID305\ESPAsyncTCPbuffer.o
Archiving .pioenvs\nodemcuv2\libFrameworkArduino.a
.piolibdeps\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp: In constructor 'AsyncClient::AsyncClient(tcp_pcb*)':
.piolibdeps\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp:77:28: error: invalid conversion from 'int8_t (*)(void*, tcp_pcb*, pbuf*, int8_t) {aka signed char (*)(void*, 

tcp_pcb*, pbuf*, signed char)}' to 'tcp_recv_fn {aka long int (*)(void*, tcp_pcb*, pbuf*, long int)}' [-fpermissive]
tcp_recv(_pcb, &_s_recv);
^

Severity | Provider | Description | Line
-- | -- | -- | --

Severity | Provider | Description | Line
-- | -- | -- | --

Severity | Provider | Description | Line
-- | -- | -- | --

thetonto, 2017-12-16:
Switch to another Git Lib and this message appears.  (seen this alot of times)
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/page/projectconf.html
 
[env:nodemcuv2]
platform = https://github.com/platformio/platform-espressif8266.git#feature/stage
board = nodemcuv2
framework = arduino
 
lib_deps = https://github.com/marvinroger/homie-esp8266.git#v2.0.0-beta.1
 

32.2 s
platformio run --target upload
If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:
 
  git checkout -b <new-branch-name>
 
Checking out files: 100% (111/111), done.
Error: Traceback (most recent call last):

  File "c:\users\the_d\.platformio\penv\lib\site-packages\platformio\__main__.py", line 120, in main
    cli(None, None, None)
  File "c:\users\the_d\.platformio\penv\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\the_d\.platformio\penv\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\users\the_d\.platformio\penv\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
Linter
Severity 	Provider 	Description	Line

ivankravets, 2017-12-16:
```ini
[env:nodemcuv2]
platform = https://github.com/platformio/platform-espressif8266.git#feature/stage
board = nodemcuv2
framework = arduino

lib_deps = 
  ArduinoJson@<5.11.0
  https://github.com/marvinroger/homie-esp8266.git#v2.0.0-beta.2
```

thetonto, 2017-12-16:
Error is now

[env:nodemcuv2]
platform = https://github.com/platformio/platform-espressif8266.git#feature/stage
board = nodemcuv2
framework = arduino
 
lib_deps =
  ArduinoJson@<5.11.0
  https://github.com/marvinroger/homie-esp8266.git#v2.0.0-beta.2
 

187.0 s
platformio run
Compiling .pioenvs\nodemcuv2\lib\ESPAsyncTCP_ID305\AsyncPrinter.o
Compiling .pioenvs\nodemcuv2\lib\ESPAsyncTCP_ID305\ESPAsyncTCP.o
Compiling .pioenvs\nodemcuv2\lib\ESPAsyncTCP_ID305\ESPAsyncTCPbuffer.o
Archiving .pioenvs\nodemcuv2\libFrameworkArduino.a
Compiling .pioenvs\nodemcuv2\lib\ESPAsyncTCP_ID305\SyncClient.o
.piolibdeps\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp: In constructor 'AsyncClient::AsyncClient(tcp_pcb*)':
.piolibdeps\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp:77:28: error: invalid conversion from 'int8_t (*)(void*, tcp_pcb*, pbuf*, int8_t) {aka signed char (*)(void*, 

tcp_pcb*, pbuf*, signed char)}' to 'tcp_recv_fn {aka long int (*)(void*, tcp_pcb*, pbuf*, long int)}' [-fpermissive]
tcp_recv(_pcb, &_s_recv);
^
In file included from .piolibdeps\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp:27:0:
Linter
Severity 	Provider 	Description	Line 

Note: this is all compiling fine over on Arduino

ivankravets, 2017-12-16:
Sorry, I corrected config above. Please use the latest HOMIE version.

thetonto, 2017-12-16:
And back to this error.    Not sure what is going on here.  All this compiles great on ArdunProject
171216-211454-nodemcuv2
lib
src
.gitignore
.travis.yml
platformio.ini
Test Homie
lib
src
.gitignore
.travis.yml
platformio.ini
platformio.ini — 171216-211454-nodemcuv2
PlatformIO Home
platformio.ini — Test Homie
1

; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/page/projectconf.html
 
[env:nodemcuv2]
platform = https://github.com/platformio/platform-espressif8266.git#feature/stage
board = nodemcuv2
framework = arduino
 
lib_deps =
  ArduinoJson@<5.11.0
  https://github.com/marvinroger/homie-esp8266.git
 

48.3 s
platformio run
  `pip install -U platformio` command
 
* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html
 
* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

 
============================================================
 
 
Linter
Severity 	Provider 	Description	Line 
PIO Buildplatformio.ini00018:51
MixedUTF-8INI0 files
C/C++ project index rebuild failed
Cloning into 'C:\Users\the_d\Documents\PlatformIO\Projects\Test Homie\.piolibdeps\_tmp_installing-ydhm4a-package'...
Checking out files:  26% (32/119)   
Checking out files:  27% (33/119)   
Checking out files:  28% (34/119)   
Checking out files:  29% (35/119)   
Checking out files:  30% (36/119)   
Checking out files:  31% (37/119)   
Checking out files:  32% (39/119)   
Checking out files:  33% (40/119)   
Checking out files:  34% (41/119)   
Checking out files:  35% (42/119)   
Checking out files:  36% (43/119)   
Checking out files:  37% (45/119)   
[... 81 lines cut ...]
    pg = ProjectGenerator(project_dir, ide, env_name)
  File "c:\users\the_d\.platformio\penv\lib\site-packages\platformio\ide\projectgenerator.py", line 38, in __init__
    self._gather_tplvars()
  File "c:\users\the_d\.platformio\penv\lib\site-packages\platformio\ide\projectgenerator.py", line 155, in _gather_tplvars
    self._tplvars.update(self.get_project_build_data())
  File "c:\users\the_d\.platformio\penv\lib\site-packages\platformio\util.py", line 134, in __call__
    value = self.func(*args)
  File "c:\users\the_d\.platformio\penv\lib\site-packages\platformio\ide\projectgenerator.py", line 79, in get_project_build_data
    raise result.exception
WindowsError: [Error 2] The system cannot find the file specified

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

ivankravets, 2017-12-16:
Please open PIO IDE Terminal and type `pio run`. Do you see any errors?

thetonto, 2017-12-16:
Same errors as before

thetonto, 2017-12-16:
.piolibdeps\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp: In constructor 'AsyncClient::AsyncClient(tcp_pcb*)':
.piolibdeps\ESPAsyncTCP_ID305\src\ESPAsyncTCP.cpp:77:28: error: invalid conversion from 'int8_t (*)(void*, tcp_pcb*, pbuf*, int8_t) {aka signed char (*)(void*,
 tcp_pcb*, pbuf*, signed char)}' to 'tcp_recv_fn {aka long int (*)(void*, tcp_pcb*, pbuf*, long int)}' [-fpermissive]
tcp_recv(_pcb, &_s_recv);

ivankravets, 2017-12-19:
Ok, it seems that issue relates to a specific library. Please explain in details here http://community.platformio.org/

closed 2017-12-19 by ivankravets