platformio/platformio-atom-ide #1276: InstallationManager

erilun06, 2017-12-17:
Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\elev\.platformio\.cache\tmp-4940a75X638lbPMh\virtualenv-14.0.6\virtualenv.py", line 2304, in <module>
    main()
  File "C:\Users\elev\.platformio\.cache\tmp-4940a75X638lbPMh\virtualenv-14.0.6\virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "C:\Users\elev\.platformio\.cache\tmp-4940a75X638lbPMh\virtualenv-14.0.6\virtualenv.py", line 924, in create_environment
    download=download,
  File "C:\Users\elev\.platformio\.cache\tmp-4940a75X638lbPMh\virtualenv-14.0.6\virtualenv.py", line 877, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env)
  File "C:\Users\elev\.platformio\.cache\tmp-4940a75X638lbPMh\virtualenv-14.0.6\virtualenv.py", line 747, in call_subprocess
    "Error %s while executing command %s" % (e, cmd_desc))
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc5 in position 16: ordinal not in range(128)

ivankravets, 2017-12-28:
@erilun06 , @weezixx 

Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

weezixx, 2017-12-28:
I don't understand why I don't have access for pip --version and pip list and why echo %PYTHONPATH% doens't work properly

C:\Users\Kevin>echo %PATH%
C:\ProgramData\Oracle\Java\javapath;C:\Program Files\ImageMagick-7.0.1-Q16;C:\Users\Kevin\AppData\Local\Programs\Python\Python35\Lib\site-packages\PyQt5;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Users\Kevin\gcc\bin;C:\Users\Kevin\gcc\libexec\gcc\x86_64-pc-mingw32\4.6.3;;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Program Files\Common Files\Autodesk Shared\;C:\Program Files (x86)\Calibre2\;C:\Program Files\MiKTeX 2.9\miktex\bin\x64\;C:\Program Files (x86)\Windows Live\Shared;C:\WINDOWS\system32\config\systemprofile\.dnx\bin;C:\Program Files\Microsoft DNX\Dnvm\;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files\Git\cmd;C:\Program Files\RogueKiller;C:\Program Files\Ghostgum\pstoedit;C:\Program Files\MiKTeX 2.9;C:\Program Files (x86)\gs\gs9.09\lib;C:\Strawberry\c\bin;C:\Strawberry\perl\site\bin;C:\Strawberry\perl\bin;C:\MinGW\bin;C:\Users\Kevin\AppData\Local\Programs\Python\Python35\Scripts\;C:\Users\Kevin\AppData\Local\Programs\Python\Python35\;C:\Users\Kevin\AppData\Local\Microsoft\WindowsApps;C:\Users\Kevin\AppData\Local\Programs\MiKTeX 2.9\miktex\bin\x64\;C:\MinGW\bin;

C:\Users\Kevin>echo %PYTHONPATH%
%PYTHONPATH%

C:\Users\Kevin>python --version
Python 3.5.1

C:\Users\Kevin>pip --version
Accès refusé.

C:\Users\Kevin>pip list
Accès refusé.

Thanks a lot again ;)

ivankravets, 2017-12-28:
@weezixx did you manually install Python 3.5.1?

weezixx, 2017-12-28:
Honestly I don't remember :/

ivankravets, 2017-12-28:
@weezixx please contact me@ikravets.com for further details. Would be thankful.

weezixx, 2017-12-28:
Isn't possible to do it here ?

ivankravets, 2017-12-28:
@weezixx I need TeamViewer session to debug this issue. I see a problem. Would be thankful if you provide TeamViewer access where I can find a proper solution to avoid a similar problem.

weezixx, 2017-12-28:
Never ^^ Sorry

ivankravets, 2017-12-28:
@weezixx I've just cleaned our thread to remove sensitive information. Please try this solution:

1. Please navigate to OS Control Panel > Applications and uninstall ALL Pythons from a system
2. Remove `%HOME_DIR%/.platformio`
3. Restart IDE.

weezixx, 2017-12-29:
Done

ivankravets, 2017-12-29:
Does it work now?

weezixx, 2017-12-29:
Nope, still got the : linter gcc executable not found and a window terminal pop up to say : access denied

weezixx, 2018-01-01:
up ? ^^

weezixx, 2018-01-01:
Ok so for the "Access denied" part, it was my antivirus -_- 

But I still have the message : linter gcc executable not found "/usr/bin/g++" not found

ivankravets, 2018-01-10:
Can you reproduce this issue with the latest version of PlatformIO IDE 2.0.0-rc.6?

ivankravets, 2018-01-10:
@weezixx please disable Antivirus for a while and repeat https://github.com/platformio/platformio-atom-ide/issues/1276#issuecomment-354364067

ivankravets, 2018-01-30:
@pewu78 do you have non-ASCII chars in the user name?

pewu78, 2018-01-30:
No, I wouldn't use them myself.

ivankravets, 2018-01-30:
@pewu78 
Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

pewu78, 2018-01-30:
I'll update on Thursday, don't have access to the machine now.

pewu78, 2018-02-01:
>echo %PATH%
C:\Python27\;C:\Python27\Scripts;C:\Program Files (x86)\NVIDIA Corporation\PhysX
\Common;C:\Program Files\ImageMagick-6.9.2-Q16;C:\Windows\system32;C:\Windows;C:
\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Program Fi
les\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\;C:\Progr
am Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\Program Files\Op
enVPN\bin;C:\Program Files\LLVM\bin;C:\Program Files\Git\cmd;C:\Program Files (x
86)\GNU\GnuPG\pub;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Common Files
\Intel\WirelessCommon\;C:\Users\Piotr\AppData\Local\atom\bin;c:\Users\Piotr\Prog
ramy\LuxRender;C:\Program Files\Microsoft VS Code\bin;C:\Program Files\Docker To
olbox
>echo %PYTHONPATH%
%PYTHONPATH%

>python --version
Python 2.7.13

>pip --version
pip 9.0.1 from c:\python27\lib\site-packages (python 2.7)

>pip list
DEPRECATION: The default format will switch to columns in the future. You can us
e --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.con
f under the [list] section) to disable this warning.
pip (9.0.1)
setuptools (28.8.0)

pewu78, 2018-02-07:
I don't have this issue anymore, in the meantime updates to newer version were applied...

ivankravets, 2018-02-07:
> updates to newer version were applied...

PIO Core? Python? PIO IDE? Which updates?

pewu78, 2018-02-07:
Python not, no env changes.
PIO IDE I guess(2.0.1), or Core (3.5.2a7), don't know if there's any history to find out.

ivankravets, 2020-06-13:
We have just released a new version [2.7.0](https://github.com/platformio/platformio-atom-ide/releases/tag/v2.7.0), please upgrade.

**Please note that we HIGHLY recommend moving to [PlatformIO IDE for VSCode](https://platformio.org/platformio-ide) which resolves tons of issues and brings a lot of new features.**

closed 2020-06-13 by ivankravets