platformio/platformio-atom-ide #1277: InstallationManager

PMorelGithub, 2017-12-17:
PIP: Traceback (most recent call last):
  File "c:\Users\Pierre\.platformio\python27\Lib\runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "c:\Users\Pierre\.platformio\python27\Lib\runpy.py", line 72, in _run_code
    exec code in run_globals
  File "C:\Users\Pierre\.platformio\penv\Scripts\pip.exe\__main__.py", line 5, in <module>
  File "c:\users\pierre\.platformio\penv\lib\site-packages\pip\__init__.py", line 21, in <module>
    from pip._vendor.requests.packages.urllib3.exceptions import DependencyWarning
  File "c:\users\pierre\.platformio\penv\lib\site-packages\pip\_vendor\requests\__init__.py", line 62, in <module>
    from .packages.urllib3.exceptions import DependencyWarning
  File "c:\users\pierre\.platformio\penv\lib\site-packages\pip\_vendor\requests\packages\__init__.py", line 29, in <module>
    import urllib3
ImportError: No module named urllib3

Windows 7

Résults for PATH PYTHONPATH python --version pip --version

C:\Users\Pierre>echo %PATH%
c:\Users\Pierre\.platformio\python27;c:\Users\Pierre\.platformio\penv\Scripts;C:
\ProgramData\Oracle\Java\javapath;C:\DEV\ADT-BUNDLE\SDK\PLATFORM-TOOLS;C:\DEV\AD
T-BUNDLE\SDK\TOOLS;C:\USERS\PIERRE\APPDATA\ROAMING\NPM;C:\DEV\APACHE-ANT-1.9.4\B
IN;C:\PROGRAM FILES\BROADCOM\BROADCOM 802.11\DRIVER;C:\PROGRAM FILES (X86)\AMD A
PP\BIN\X86_64;C:\PROGRAM FILES (X86)\AMD APP\BIN\X86;C:\PROGRAM FILES\COMMON FIL
ES\MICROSOFT SHARED\WINDOWS LIVE;C:\PROGRAM FILES (X86)\COMMON FILES\MICROSOFT S
HARED\WINDOWS LIVE;C:\WINDOWS\SYSTEM32;C:\WINDOWS;C:\WINDOWS\SYSTEM32\WBEM;C:\WI
NDOWS\SYSTEM32\WINDOWSPOWERSHELL\V1.0\;C:\PROGRAM FILES (X86)\WINDOWS LIVE\SHARE
D;C:\PROGRAM FILES (X86)\ATI TECHNOLOGIES\ATI.ACE\CORE-STATIC;C:\PROGRAM FILES\W
IDCOMM\BLUETOOTH SOFTWARE\;C:\PROGRAM FILES\WIDCOMM\BLUETOOTH SOFTWARE\SYSWOW64;
C:\PROGRAM FILES\JAVA\JDK1.8.0_11\BIN;C:\PROGRAM FILES\MICROSOFT\WEB PLATFORM IN
STALLER\;C:\PROGRAM FILES (X86)\GIT\CMD;C:\PROGRAM FILES (X86)\MICROSOFT ASP.NET
\ASP.NET WEB PAGES\V1.0\;C:\PROGRAM FILES\MICROSOFT SQL SERVER\110\TOOLS\BINN\;C
:\PROGRAM FILES (X86)\MICROSOFT SQL SERVER\100\TOOLS\BINN\;C:\PROGRAM FILES\MICR
OSOFT SQL SERVER\100\TOOLS\BINN\;C:\PROGRAM FILES\MICROSOFT SQL SERVER\100\DTS\B
INN\;C:\PROGRAM FILES (X86)\WINDOWS KITS\8.0\WINDOWS PERFORMANCE TOOLKIT\;C:\PRO
GRAM FILES (X86)\VISUALSVN SERVER\BIN;C:\PROGRAM FILES\TORTOISESVN\BIN;C:\PROGRA
M FILES (X86)\VISUALSVN\BIN;C:\PROGRAM FILES\TORTOISEGIT\BIN;C:\Program Files (x
86)\QuickTime\QTSystem\;C:\Program Files (x86)\Skype\Phone\;C:\Program Files\PuT
TY\;C:\Program Files\nodejs\;C:\Program Files (x86)\Windows Kits\8.1\Windows Per
formance Toolkit\;C:\Program Files\LLVM\bin;C:\Users\Pierre\AppData\Local\.meteo
r\;C:\Program Files (x86)\Scapple;C:\Users\Pierre\AppData\Roaming\npm;C:\Users\P
ierre\AppData\Local\atom\bin

C:\Users\Pierre>echo %PYTHONPATH%
c:\Users\Pierre\.platformio\python27\Lib;c:\Users\Pierre\.platformio\python27\To
ols;

C:\Users\Pierre>python --version
Python 2.7.14

C:\Users\Pierre>pip --version
Traceback (most recent call last):
  File "c:\Users\Pierre\.platformio\python27\Lib\runpy.py", line 174, in _run_mo
dule_as_main
    "__main__", fname, loader, pkg_name)
  File "c:\Users\Pierre\.platformio\python27\Lib\runpy.py", line 72, in _run_cod
e
    exec code in run_globals
  File "c:\Users\Pierre\.platformio\penv\Scripts\pip.exe\__main__.py", line 5, i
n <module>
  File "c:\users\pierre\.platformio\penv\lib\site-packages\pip\__init__.py", lin
e 21, in <module>
    from pip._vendor.requests.packages.urllib3.exceptions import DependencyWarni
ng
  File "c:\users\pierre\.platformio\penv\lib\site-packages\pip\_vendor\requests\
__init__.py", line 62, in <module>
    from .packages.urllib3.exceptions import DependencyWarning
  File "c:\users\pierre\.platformio\penv\lib\site-packages\pip\_vendor\requests\
packages\__init__.py", line 29, in <module>
    import urllib3
ImportError: No module named urllib3

I uninstall all software having python.dll (12 softwares) A gloal search show that no more python.dll exist on my computer, only the one in .platformio

Thanks for the help

ivankravets, 2017-12-19:
Could you try this solution https://github.com/platformio/platformio-atom-ide/issues/1279#issuecomment-352708612 ?

ivankravets, 2018-01-10:
Please reopen if you still need help

closed 2018-01-10 by ivankravets

seanrm5, 2018-10-31:
Hello guys i got the same error here while trying to make work a instabot…
I really don't understand how to solve this problem
Thank you in advance!!!

[image]