platformio/platformio-atom-ide #1279: InstallationManager

PMorelGithub, 2017-12-18:
I try to install package platformio_ide and I receive errors. I am on Windows 7

I do installation and deinstallation of Atom, after deleting the .atom and the .platformio directories
I have set the PATH and the PYTHONPATH with forward and back slash and also double back slash and nothing resolve my problems.
I already have deinstall all the python using python having python dll in their file tree

Here is the trace:
Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\Pierre\.platformio\.cache\tmp-5452MKKI099w83BK\virtualenv-14.0.6\virtualenv.py", line 2304, in <module>
    main()
  File "C:\Users\Pierre\.platformio\.cache\tmp-5452MKKI099w83BK\virtualenv-14.0.6\virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "C:\Users\Pierre\.platformio\.cache\tmp-5452MKKI099w83BK\virtualenv-14.0.6\virtualenv.py", line 904, in create_environment
    site_packages=site_packages, clear=clear, symlink=symlink))
  File "C:\Users\Pierre\.platformio\.cache\tmp-5452MKKI099w83BK\virtualenv-14.0.6\virtualenv.py", line 1101, in install_python
    copy_required_modules(home_dir, symlink)
  File "C:\Users\Pierre\.platformio\.cache\tmp-5452MKKI099w83BK\virtualenv-14.0.6\virtualenv.py", line 1041, in copy_required_modules
    dst_filename = change_prefix(filename, dst_prefix)
  File "C:\Users\Pierre\.platformio\.cache\tmp-5452MKKI099w83BK\virtualenv-14.0.6\virtualenv.py", line 1015, in change_prefix
    (filename, prefixes)
AssertionError: Filename C:\python27\Lib\os.py does not start with any of these prefixes: ['C:\\Python27']

Here is my PATH, PYTHONPATH, python --version pip --version
Microsoft Windows [version 6.1.7601]
Copyright (c) 2009 Microsoft Corporation. Tous droits réservés.

C:\Windows\system32>echo %PATH%
C:\\Python27\\;C:\\Python27\\Scripts\\;C:\ProgramData\Oracle\Java\javapath;C:\DE
V\ADT-BUNDLE\SDK\PLATFORM-TOOLS;C:\DEV\ADT-BUNDLE\SDK\TOOLS;C:\USERS\PIERRE\APPD
ATA\ROAMING\NPM;C:\DEV\APACHE-ANT-1.9.4\BIN;C:\PROGRAM FILES\BROADCOM\BROADCOM 8
02.11\DRIVER;C:\PROGRAM FILES (X86)\AMD APP\BIN\X86_64;C:\PROGRAM FILES (X86)\AM
D APP\BIN\X86;C:\PROGRAM FILES\COMMON FILES\MICROSOFT SHARED\WINDOWS LIVE;C:\PRO
GRAM FILES (X86)\COMMON FILES\MICROSOFT SHARED\WINDOWS LIVE;C:\WINDOWS\SYSTEM32;
C:\WINDOWS;C:\WINDOWS\SYSTEM32\WBEM;C:\WINDOWS\SYSTEM32\WINDOWSPOWERSHELL\V1.0\;
C:\PROGRAM FILES (X86)\WINDOWS LIVE\SHARED;C:\PROGRAM FILES (X86)\ATI TECHNOLOGI
ES\ATI.ACE\CORE-STATIC;C:\PROGRAM FILES\WIDCOMM\BLUETOOTH SOFTWARE\;C:\PROGRAM F
ILES\WIDCOMM\BLUETOOTH SOFTWARE\SYSWOW64;C:\PROGRAM FILES\JAVA\JDK1.8.0_11\BIN;C
:\PROGRAM FILES\MICROSOFT\WEB PLATFORM INSTALLER\;C:\PROGRAM FILES (X86)\GIT\CMD
;C:\PROGRAM FILES (X86)\MICROSOFT ASP.NET\ASP.NET WEB PAGES\V1.0\;C:\PROGRAM FIL
ES\MICROSOFT SQL SERVER\110\TOOLS\BINN\;C:\PROGRAM FILES (X86)\MICROSOFT SQL SER
VER\100\TOOLS\BINN\;C:\PROGRAM FILES\MICROSOFT SQL SERVER\100\TOOLS\BINN\;C:\PRO
GRAM FILES\MICROSOFT SQL SERVER\100\DTS\BINN\;C:\PROGRAM FILES (X86)\WINDOWS KIT
S\8.0\WINDOWS PERFORMANCE TOOLKIT\;C:\PROGRAM FILES (X86)\VISUALSVN SERVER\BIN;C
:\PROGRAM FILES\TORTOISESVN\BIN;C:\PROGRAM FILES (X86)\VISUALSVN\BIN;C:\PROGRAM
FILES\TORTOISEGIT\BIN;C:\Program Files (x86)\QuickTime\QTSystem\;C:\Program File
s (x86)\Skype\Phone\;C:\Program Files\PuTTY\;C:\Program Files\nodejs\;C:\Program
 Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\Program Files\LLVM
\bin;C:\Users\Pierre\AppData\Local\.meteor\;C:\Program Files (x86)\Scapple;C:\Us
ers\Pierre\AppData\Roaming\npm;C:\Users\Pierre\AppData\Local\atom\bin

C:\Windows\system32>echo %PYTHONPATH%
c:\\python27\\Lib;c:\\python27\\Tools;

C:\Windows\system32>python --version
Python 2.7.14

C:\Windows\system32>pip --version
pip 9.0.1 from c:\python27\lib\site-packages (python 2.7)

C:\Windows\system32>pip list
DEPRECATION: The default format will switch to columns in the future. You can us
e --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.con
f under the [list] section) to disable this warning.
pip (9.0.1)
setuptools (28.8.0)

C:\Windows\system32>

ivankravets, 2017-12-19:
1. Please navigate to OS Control Panel > Applications and uninstall ALL Pythons from a system
2. Remove `%HOME_DIR%/.platformio`
3. Restart IDE.

Now all should work.

PMorelGithub, 2017-12-19:
I read and I made this proposal to solve the problem more than a dozen times. I even searched for all python.dll files on my hard drive and then uninstalled these applications, but nothing works.
This software seems to be a very nice development platform, but it is useless if we can not get it started.

Is there a specific sequence of setter PATH and PYTHONPATH, it is not explain. Are there many values to check in the Registry? I am in Windows 7. There must be a solution. Sorry to let my frustration appear, but it's that I'd like it to work and use this fantastic software.

ivankravets, 2017-12-19:
After removing ALL Pythons from a system, cleaning env PATH, and removing `c:\python27`, do you see the same error?

PMorelGithub, 2017-12-19:
Thanks for the help, but it didn't solve the problem. I give up.

PMorelGithub, 2017-12-21:
It is working at the end. What I have done.
Prior to the last action, I was installing the Windows Intsaller python-2.7.14.amd64.msi 
I decide to use the 32bit installation, just for a try, the python-2.7.14.msi
I select in the installation to add it to the Path.
I do a Cmd Prompt test as an Administrator and doing python --version work. 
I do a Cmd Prompt test without being Administrator, only a user and doing python --version gave a program not found error
So I add to my user Path the C:/Python27;C:/Python27/Scripts  
Doing a pyhon --version in a Cmd Prompt as a user gave me the version number so it was working
I deinstall Atom. I erase the .atom and .platformio from my user directory in my case C:/Users/Pierre
I install Atom and in Atom settings-Install, I select the plateformio_ide package. When finish, I do a restart of Atom.
The program goes until it was the loading of the Platform IO Home. I see a spinning wheel and I let it goes for a very long time, but whithout viewing errors....
I select the menu PlatformIO - Update Platforms-pacage and libraries
A command windows open ans for the first time, I saw the 
contrib-piohome   0.5.3 in the list of packages, the two others are (tool-pioplus and pysite-pioplus)
I the try a pio home and voilà, the browser lunch the pio Home
And exit Atom and start it again. This time, the Platform IO Home open inside de Atom interface, and all that without those infamous errors.
I hope that my path to solve this may help others...
I can start to use this great sotware...

closed 2018-01-10 by ivankravets