platformio/platformio-atom-ide #1282: InstallationManager

Ertany, 2017-12-19:
PIP:   Failed building wheel for netifaces
Command "/home/ethan/.platformio/penv/bin/python2.7 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-WDZuh3/netifaces/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-Xolplq-record/install-record.txt --single-version-externally-managed --compile --install-headers /home/ethan/.platformio/penv/include/site/python2.7/netifaces" failed with error code 1 in /tmp/pip-build-WDZuh3/netifaces/

ivankravets, 2017-12-19:
What is your OS?

ivankravets, 2017-12-19:
Please restart IDE, fixed in
https://github.com/platformio/platformio-core/commit/c6abdf8206b4ed156bdffb6cd481d1eda8e12250

closed 2017-12-19 by ivankravets