platformio/platformio-atom-ide #1283: InstallationManager

juankyestevez, 2017-12-19:
PIP:   Failed building wheel for netifaces
Command "/home/juanky/.platformio/penv/bin/python2.7 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-Awm95Q/netifaces/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-1MSaMd-record/install-record.txt --single-version-externally-managed --compile --install-headers /home/juanky/.platformio/penv/include/site/python2.7/netifaces" failed with error code 1 in /tmp/pip-build-Awm95Q/netifaces/

ivankravets, 2017-12-19:
What is your OS?

ivankravets, 2017-12-19:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/1282

closed 2017-12-19 by ivankravets

juankyestevez, 2017-12-19:
Hi, I am testing Deepin 15.5

ivankravets, 2017-12-19:
Please restart IDE. All should work now.

juankyestevez, 2017-12-19:
"PlatformIO IDE has been successfully installed!
Please restart Atom to apply the changes"

After this message I restart Atom, but after start atom platformIO is installed again.

[image]