platformio/platformio-atom-ide #1284: InstallationManager

jrlortiz, 2017-12-19:
PIP:   Failed building wheel for netifaces
Command "/home/openhabuser/.platformio/penv/bin/python2.7 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-hMAznH/netifaces/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-pLO_hb-record/install-record.txt --single-version-externally-managed --compile --install-headers /home/openhabuser/.platformio/penv/include/site/python2.7/netifaces" failed with error code 1 in /tmp/pip-build-hMAznH/netifaces/

ivankravets, 2017-12-19:
What is your OS?

ivankravets, 2017-12-19:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/1282

closed 2017-12-19 by ivankravets