platformio/platformio-atom-ide #1286: PlatformIO Core keeps installing after every restart

aelesia, 2017-12-19:
Every single time I boot Atom, I get a prompt saying that PlatformIO Core is downloading and installing, and to reboot it after it's done. However after I reboot it, the same thing happens again and again.

I saw this ticket here: https://github.com/platformio/platformio-atom-ide/issues/55

Tried renaming /build/ and /platformio-ide/, but upon restart, it simply says that it's redownloading and installing dependencies, and it re-creates the file again, then the same issue happens.

wurmtal868, 2017-12-19:
Same for me.
I tried removing and deleting ~/.platformio 
After re-installing it's still restarting

ivankravets, 2017-12-19:
Sorry, should be fixed now. Just restart IDE the last time.

closed 2017-12-19 by ivankravets

jaslyn-s, 2017-12-19:
Thanks. It's fixed.