platformio/platformio-atom-ide #1294: C/C++ project index rebuild failed

9a4gl, 2017-12-21:
Error: [Thu Dec 21 11:40:27 2017] Processing heltek_kit_8 (platform: espressif8266; board: heltek_kit_8; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Error: Unknown board ID 'heltek_kit_8'
========================================================================================= [ERROR] Took 0.19 seconds =========================================================================================

9a4gl, 2017-12-21:
This looks like misspelling, there is heltec_wifi_kit_32 platform id which looks ok.
This one should be called heltec_wifi_kit_8, note the company is heltec, not heltek

9a4gl, 2017-12-21:
Looks like heltek_kit_8 has been added 5 days ago
https://github.com/platformio/platform-espressif8266/commit/fa3cbbee36565f5ea873586682f12f53a1ac1f18#diff-6c437de2a5ca04d57fa70305850c2b7e

We need new release of platformio/platform-espressif8266 please

ivankravets, 2017-12-22:
We will merge soon. Please re-test with http://docs.platformio.org/en/latest/platforms/espressif8266.html#using-arduino-framework-with-staging-version

closed 2017-12-22 by ivankravets