platformio/platformio-atom-ide #1298: _remove_dead_weakref does not exist

pdietrich, 2017-12-23:
I'm having the following problem when I try to access the PIO home screen. I'm using Windows 7. It happens with a fresh Atom installation as well as with a fresh VS Code installation. I have already removed all other previously installed Python distributions.

I can also reproduce it on the command line:
```
PS C:\Users\Patrick\.platformio\penv\Scripts> .\pio home
Traceback (most recent call last):
  File "main.py", line 12, in init main
  File "c:\users\patrick\.platformio\penv\lib\site-packages\click\__init__.py", line 18, in <module>
    from .core import Context, BaseCommand, Command, MultiCommand, Group, \
  File "c:\users\patrick\.platformio\penv\lib\site-packages\click\core.py", line 8, in <module>
    from .types import convert_type, IntRange, BOOL
  File "c:\users\patrick\.platformio\penv\lib\site-packages\click\types.py", line 5, in <module>
    from ._compat import open_stream, text_type, filename_to_ui, \
  File "c:\users\patrick\.platformio\penv\lib\site-packages\click\_compat.py", line 6, in <module>
    from weakref import WeakKeyDictionary
  File "C:\Users\Patrick\.platformio\python27\Lib\weakref.py", line 14, in <module>
    from _weakref import (
ImportError: cannot import name _remove_dead_weakref
```

ivankravets, 2017-12-23:
Could try [this solution](https://community.platformio.org/t/cant-get-pio-to-work-home-doesnt-want-to-load/2755/15?u=ivankravets)?

pdietrich, 2017-12-26:
Unfortunately, that solution does not work for me; I removed the .platformio folder, then rebooted the computer (just to make sure) and restarted Atom. Atom then reinstalled PlatformIO. I then restarted Atom. After the restart I get the same Problem again.

Could you provide me with some more information on the problem? To me it seems like the python installation in ....\.platformio\python27 is broken or outdated.

pdietrich, 2017-12-26:
Ok, I have found a solution to my problem.
I used regedit to delete the following two folders from the Windows registry: HKEY_LOCAL_MACHINE/SOFTWARE/Python
HKEY_CURRENT_USER/SOFTWARE/Python

The I searched my home folder for "python" and deleted the following folders which seemed to contain their own python installations are were somehow associated with python:
C:\Users\Patrick\.ipython/
C:\Users\Patrick\AppData\Local\pip
C:\Users\Patrick\AppData\Local\NodeBox (contains python installation)
C:\Users\Patrick\AppData\Local\Temp (contained a python folder)

I then removed the C:\Users\Patrick\.platformio folder (as suggested), rebooted and restarted atom.
Now it works.

pdietrich, 2017-12-26:
So I got it working. But I would not consider this problem solved for the following reason.
It is not uncommon for developers to have other python installations. So I think PlatformIO should cope with that. I am aware that this is mainly a python problem, not a PIO problem.

Is it possible to make a python installation ignore any other python installations on the system, and in particular any registry entries and environment variables? If so doing that with the python installation in .platformio would be the right solution imho.

ivankravets, 2018-02-17:
@jmdearras

Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

jmdearras, 2018-02-17:
PS C:\Users\Jim De Aras> echo %PATH%

>>
%PATH%
PS C:\Users\Jim De Aras> echo %PYTHONPATH%
>>
%PYTHONPATH%
PS C:\Users\Jim De Aras> python --version
>>
Python 2.7.14
PS C:\Users\Jim De Aras> pip --version
>>
pip 9.0.1 from c:\users\jimdea~2\platfo~1\penv\lib\site-packages (python
2.7)
PS C:\Users\Jim De Aras> pip listDEPRECATION: The default format will
switch to columns in the future. You can use --format=(legacy|columns) (or
define a format=(legacy|columns) in your pip.conf under the [list] section)
to disable this warning.
bottle (0.12.13)
certifi (2018.1.18)
chardet (3.0.4)
click (5.1)
colorama (0.3.9)
idna (2.6)
lockfile (0.12.2)
pip (9.0.1)
platformio (3.5.2b4)
pyserial (3.4)
requests (2.18.4)
semantic-version (2.6.0)
setuptools (38.5.1)
urllib3 (1.22)
wheel (0.30.0)
PS C:\Users\Jim De Aras>

ivankravets, 2018-02-17:
please use `cmd`, not PS

jmdearras, 2018-02-17:
Microsoft Windows [Version 10.0.16299.248]
(c) 2017 Microsoft Corporation. All rights reserved.

C:\Users\Jim De Aras>echo %PATH%
C:\Program Files (x86)\Intel\iCLS Client\;C:\Program Files\Intel\iCLS
Client\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program
Files\Microsoft\Web Platform Installer\;C:\Program Files (x86)\Bitvise SSH
Client;C:\Program Files (x86)\Microsoft SQL Server\Client
SDK\ODBC\110\Tools\Binn\;C:\Program Files (x86)\Microsoft SQL
Server\120\Tools\Binn\ManagementStudio\;C:\Program Files (x86)\Microsoft
SQL Server\120\Tools\Binn\;C:\Program Files (x86)\Microsoft SQL
Server\120\DTS\Binn\;C:\Program Files (x86)\MySQL\MySQL
Utilities\;C:\Program Files (x86)\MySQL\MySQL Utilities\Doctrine extensions
for PHP\;C:\Program Files (x86)\Intel\Intel(R) Management Engine
Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine
Components\DAL;C:\Program Files (x86)\Intel\Intel(R) Management Engine
Components\IPT;C:\Program Files\Intel\Intel(R) Management Engine
Components\IPT;C:\Program
Files\Amazon\AWSCLI\;C:\WINDOWS\system32\config\systemprofile\.dnx\bin;C:\Program
Files\Microsoft DNX\Dnvm\;C:\Program Files\Microsoft SQL
Server\130\Tools\Binn\;C:\Program Files (x86)\Windows Kits\8.1\Windows
Performance Toolkit\;C:\Program Files (x86)\Skype\Phone\;C:\Program Files
(x86)\WinMerge;C:\Program Files (x86)\IVI
Foundation\VISA\WinNT\Bin;C:\Program Files\Git\cmd;C:\Program
Files\dotnet\;C:\Program Files\TortoiseGit\bin;C:\Program
Files\Intel\WiFi\bin\;C:\Program Files\Common
Files\Intel\WirelessCommon\;C:\Program Files (x86)\Microsoft SQL
Server\Client SDK\ODBC\130\Tools\Binn\;C:\Program Files (x86)\Microsoft SQL
Server\140\Tools\Binn\;C:\Program Files (x86)\Microsoft SQL
Server\140\DTS\Binn\;C:\Program Files (x86)\Microsoft SQL
Server\140\Tools\Binn\ManagementStudio\;C:\Program
Files\LLVM\bin;C:\Users\Jim De
Aras\AppData\Local\Microsoft\WindowsApps;C:\Users\Jim De
Aras\AppData\Local\Google\Cloud SDK\google-cloud-sdk\bin;C:\Users\Jim De
Aras\AppData\Local\GitHubDesktop\bin;C:\Users\Jim De
Aras\AppData\Local\Microsoft\WindowsApps;C:\Program
Files\Intel\WiFi\bin\;C:\Program Files\Common
Files\Intel\WirelessCommon\;C:\Users\Jim De Aras\AppData\Local\atom\bin

C:\Users\Jim De Aras>echo %PYTHONPATH%
%PYTHONPATH%

C:\Users\Jim De Aras>python --version
'python' is not recognized as an internal or external command,
operable program or batch file.

C:\Users\Jim De Aras>pip --version
'pip' is not recognized as an internal or external command,
operable program or batch file.

C:\Users\Jim De Aras>pip list
'pip' is not recognized as an internal or external command,
operable program or batch file.

sovcik, 2018-05-21:
The same issue here. Including system variables.

sovcik, 2018-05-22:
Not sure how/if that registry editing helped, but removing .platformio did the thing. But had to kill several pioplus.exe and python.exe processes. Python.exe was started from .pltaftomio folder.

ivankravets, 2018-05-24:
@sovcik 
Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

sovcik, 2018-05-25:
`````
C:\Users\Jozef>echo %path%
C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\ProgramData\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files\TortoiseSVN\bin;C:\Program Files (x86)\GtkSharp\2.12\bin;C:\ProgramData\chocolatey\bin;C:\Program Files\Git\cmd;C:\Program Files (x86)\eID klient\;C:\Program Files\nodejs\;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\;C:\Program Files\Java\jdk1.8.0_92\bin;C:\Users\Jozef\Apps\apache-ant-1.10.1\bin;C:\Users\Jozef\Apps\apache-maven-3.3.9\bin;C:\Users\Jozef\AppData\Local\Microsoft\WindowsApps;C:\Program Files\Heroku\bin;C:\Program Files\MongoDB\Server\3.4\bin;C:\Users\Jozef\AppData\Roaming\npm;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\;;C:\Program Files\Microsoft VS Code\bin

C:\Users\Jozef>echo %pytonpath%
%pytonpath%

C:\Users\Jozef>python --version
'python' is not recognized as an internal or external command,
operable program or batch file.

C:\Users\Jozef>pip --version
'pip' is not recognized as an internal or external command,
operable program or batch file.

`````

As I mentioned - it started working after I 
1. removed Python from registry 
1. removed that `.platformio` folder

But I'm not sure if removing Python from registry was necessary - I just followed recommendation mentioned earlier in the thread.

Another symptom of that issues was that `PIO Home` did not work either. It tried to load news, but never finished. But it works now too.

And just noticed that this issue is related to AtomIDE... I'm using VS Code... Maybe my comments are irrelevant. Sorry for mixing it up...

ivankravets, 2018-06-19:
Should be fixed in the latest release

closed 2018-06-19 by ivankravets

atmjoy, 2018-07-27:
hello, apparently pb with python, works with 2.7.13, but not with 2.7.14 or 2.7.15

ivankravets, 2018-07-27:
@atmjoy we have just redeployed new PIO Plus Core which was built using Python 2.7.15. Could you re-test the next?

1. Close Atom
2. Please uninstall all Pythons from a system, see http://bit.ly/error-dead-weakref
3. Please remove `C:\Users\YourUsername\.platformio\packages\tool-pioplus` folder
4. Please do not install Python manually, let's check how PlatformIO will do that
5. Open Atom. It should install the latest Python + PIO Core automatically.
6. Restart Atom.

Does it work now?