platformio/platformio-atom-ide #1300: C/C++ project index rebuild failed

Jeffrihr, 2017-12-23:
Error: 
[12/23/17 22:09:43] Processing genericSTM32F103CB (platform: ststm32; build_flags: -std=c++11; board: genericSTM32F103CB; framework: arduino)

Verbose mode can be enabled via `-v, --verbose` option
Converting OpenPRO58.ino
Collected 30 compatible libraries
Looking for dependencies...
Library Dependency Graph ( http://bit.ly/configure-pio-ldf )
|-- <Wire>
|-- <Adafruit_SSD1360_STM32>
|   |-- <Wire>
|   |-- <Adafruit_GFX_AS>
|   |   |-- <Adafruit_GFX_AS>
|-- <Adafruit_GFX_AS>
|   |-- <Adafruit_GFX_AS>
|-- <Eeprom24C01_02>
|   |-- <Wire>

*** 'B:\Desktop\OpenPRO58-master\OpenPRO58\.pioenvs\genericSTM32F103CB\lib\Adafruit_GFX_AS' already has a source directory: 'C:\users\Jeffri Hafiz Rahmat\.platformio\packages\framework-arduinoststm32\STM32F1\libraries\Adafruit_GFX_AS'.
File "C:\users\jeffri~1\platfo~1\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 229, in CollectBuildFiles
 [ERROR] Took 1.38 seconds 

 [SUMMARY] 
Environment genericSTM32F103C8	[SKIP]
Environment genericSTM32F103CB	[ERROR]
 [ERROR] Took 1.38 seconds

ivankravets, 2017-12-27:
Thanks! Fixed in https://github.com/platformio/platformio-core/commit/012cb85e31525c925cac48013a92635b8cff58d5

Please open PIO IDE Terminal and type `pio upgrade --dev`

closed 2017-12-27 by ivankravets

Jeffrihr, 2017-12-28:
Thanks for the answer
I'll try it
I'm noob in this hobby, sorry about my question...