platformio/platformio-atom-ide #1301: Uncaught Error: Could not start PIO Home server: Error: Traceback (most recent call last):  File...

alexfederlin, 2017-12-23:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.23.1 x64
**Electron**: 1.6.15
**OS**: Ubuntu 16.04.3
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 2.0.0-rc.4

### Stack Trace

Uncaught Error: Could not start PIO Home server: Error: Traceback (most recent call last):
  File "main.py", line 128, in main.main
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "cmd_home.py", line 509, in cmd_home.cli
  File "/usr/local/lib/python2.7/dist-packages/platformio/managers/core.py", line 46, in get_core_package_dir
    assert name in CORE_PACKAGES
AssertionError

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to support@pioplus.com

============================================================

```
At /home/alex/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1571

Error: Could not start PIO Home server: Error: Traceback (most recent call last):
  File "main.py", line 128, in main.main
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "cmd_home.py", line 509, in cmd_home.cli
  File "/usr/local/lib/python2.7/dist-packages/platformio/managers/core.py", line 46, in get_core_package_dir
    assert name in CORE_PACKAGES
AssertionError

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to support@pioplus.com

============================================================

    at /packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1571:23
    at ChildProcess.onExit (/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:499:5)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at maybeClose (internal/child_process.js:885:16)
    at Socket.<anonymous> (internal/child_process.js:334:11)
    at emitOne (events.js:96:13)
    at Socket.emit (events.js:188:7)
    at Pipe._handle.close [as _onclose] (net.js:501:12)
```

### Commands

```
     -0:17.9.0 platformio-ide:home (atom-pane.pane.active)
```

### Non-Core Packages

```
autocomplete-clang 0.11.4 
build 0.70.0 
busy-signal 1.4.3 
file-icons 2.1.15 
intentions 1.1.5 
language-ini 1.19.0 
linter 2.2.0 
linter-gcc 0.7.1 
linter-ui-default 1.6.10 
platformio-ide 2.0.0-rc.4 
platformio-ide-debugger 1.2.4 
platformio-ide-terminal 2.7.0 
tool-bar 1.1.0 
```

ivankravets, 2017-12-27:
Please open PIO IDE Terminal and type `pio upgrade`. Restart IDE.

cyrusbuilt, 2018-01-02:
I did a 'pio upgrade' and still get this error.

Output from 'pio upgrade': 
You're up-to-date!
PlatformIO 3.5.1a1 is currently the newest version available.

Error report:

[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.23.1 x64
**Electron**: 1.6.15
**OS**: Mac OS X 10.13.2
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 2.0.0-rc.5

### Stack Trace

Uncaught Error: Could not start PIO Home server: Error: Traceback (most recent call last):
  File "main.py", line 128, in main.main
  File "/Library/Python/2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/Library/Python/2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/Library/Python/2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Library/Python/2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Library/Python/2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "cmd_home.py", line 509, in cmd_home.cli
  File "/Library/Python/2.7/site-packages/platformio/managers/core.py", line 74, in get_core_package_dir
    assert name in CORE_PACKAGES
AssertionError

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to support@pioplus.com

============================================================

```
At /Users/sdsdev/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1571

Error: Could not start PIO Home server: Error: Traceback (most recent call last):
  File "main.py", line 128, in main.main
  File "/Library/Python/2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/Library/Python/2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/Library/Python/2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Library/Python/2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Library/Python/2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "cmd_home.py", line 509, in cmd_home.cli
  File "/Library/Python/2.7/site-packages/platformio/managers/core.py", line 74, in get_core_package_dir
    assert name in CORE_PACKAGES
AssertionError

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to support@pioplus.com

============================================================

    at /packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1571:23
    at ChildProcess.onExit (/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:499:5)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at maybeClose (internal/child_process.js:885:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
```

### Commands

```
```

### Non-Core Packages

```
atom-easy-jsdoc 4.12.4 
atom-typescript 11.0.10 
autocomplete-clang 0.11.4 
autocomplete-swift 1.0.3 
bottom-dock 0.4.4 
build 0.70.0 
busy 0.7.0 
busy-signal 1.4.3 
docblockr 0.13.4 
file-icons 2.1.15 
file-type-icons 1.3.4 
intentions 1.1.5 
jshint 1.8.6 
jslint 1.5.1 
language-batch 0.4.0 
language-ini 1.19.0 
language-swift 0.5.0 
language-swift-3 0.4.0 
linter 2.2.0 
linter-flake8 2.3.0 
linter-gcc 0.7.1 
linter-lightning 0.8.0 
linter-manager 0.2.12 
linter-swift-package-manager 0.1.0 
linter-swiftc 2.0.0 
linter-ui-default 1.6.10 
MavensMate-Atom 1.1.0 
minimap 4.29.7 
node-requirer 0.1.12 
platformio-ide 2.0.0-rc.5 
platformio-ide-debugger 1.2.5 
platformio-ide-terminal 2.8.0 
pretty-json 1.6.4 
python-nosetests 0.2.0 
python-tools 0.6.9 
swift-debugger 0.1.2 
todo-manager 0.2.10 
tool-bar 1.1.0 
```

ivankravets, 2018-01-02:
@cyrusbuilt you have a lot of different PIO Cores in a system. http://docs.platformio.org/en/latest/faq.html#multiple-pio-cores-in-a-system

closed 2018-01-02 by ivankravets