platformio/platformio-atom-ide #1302: Uncaught Error: Could not start PIO Home server: Please manually remove file `~/.platformio/packa...

jhsis, 2017-12-24:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.23.1 x64
**Electron**: 1.6.15
**OS**: Mac OS X 10.13.2
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 2.0.0-rc.4

### Stack Trace

Uncaught Error: Could not start PIO Home server: Please manually remove file `/Users/jhs/.platformio/packages/contrib-pysite/twisted/internet`
Please manually remove file `/Users/jhs/.platformio/packages/contrib-pysite/twisted/internet`
Error: Could not install 'contrib-pysite' with version requirements '>=0.1.1,<2' for your system 'darwin_x86_64'.
 If you use Antivirus, it can block PlatformIO Package Manager. Try to disable it for a while.

```
At /Users/jhs/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1571

Error: Could not start PIO Home server: Please manually remove file `/Users/jhs/.platformio/packages/contrib-pysite/twisted/internet`
Please manually remove file `/Users/jhs/.platformio/packages/contrib-pysite/twisted/internet`
Error: Could not install 'contrib-pysite' with version requirements '>=0.1.1,<2' for your system 'darwin_x86_64'.
 If you use Antivirus, it can block PlatformIO Package Manager. Try to disable it for a while.

    at /packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1571:23
    at ChildProcess.onExit (/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:499:5)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at maybeClose (internal/child_process.js:885:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
```

### Commands

```
```

### Non-Core Packages

```
autocomplete-clang 0.11.4 
build 0.69.0 
busy 0.7.0 
busy-signal 1.4.3 
dbg 1.6.3 
dbg-gdb 1.7.8 
file-icons 2.1.15 
intentions 1.1.5 
language-ini 1.19.0 
linter 2.2.0 
linter-gcc 0.7.1 
linter-ui-default 1.6.10 
minimap 4.29.7 
output-panel 0.3.4 
platformio-ide 2.0.0-rc.4 
platformio-ide-debugger 1.2.4 
platformio-ide-terminal 2.7.0 
tool-bar 1.1.0 
```

ivankravets, 2017-12-27:
Please remove `/Users/jhs/.platformio/packages/contrib-pysite` folder, restart IDE.

closed 2017-12-27 by ivankravets

jhsis, 2017-12-29:
Thanks Ivan,
There are many, much larger organizations that could learn customer support from your example…
Merry Christmas
Thanks again

> On Dec 27, 2017, at 12:06 PM, Ivan Kravets <notifications@github.com> wrote:
> 
> /Users/jhs/.platformio/packages/contrib-pysite

ivankravets, 2017-12-29:
Thanks! ;)

Merry Christmas!