platformio/platformio-atom-ide #1304: C/C++ project index rebuild failed

jwkennedy60, 2017-12-25:
Traceback (most recent call last):
  File "C:\Users\jwken\.platformio\penv\Scripts\platformio-script.py", line 6, in <module>
    from pkg_resources import load_entry_point
  File "C:\Python27\lib\site-packages\pkg_resources\__init__.py", line 3017, in <module>
    @_call_aside
  File "C:\Python27\lib\site-packages\pkg_resources\__init__.py", line 3003, in _call_aside
    f(*args, **kwargs)
  File "C:\Python27\lib\site-packages\pkg_resources\__init__.py", line 3030, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "C:\Python27\lib\site-packages\pkg_resources\__init__.py", line 659, in _build_master
    ws.require(__requires__)
  File "C:\Python27\lib\site-packages\pkg_resources\__init__.py", line 967, in require
    needed = self.resolve(parse_requirements(requirements))
  File "C:\Python27\lib\site-packages\pkg_resources\__init__.py", line 853, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'platformio==3.5.0rc5' distribution was not found and is required by the application

ivankravets, 2017-12-27:
Please remove `%USER_HOME%\.platformio\penv` folder and restart IDE.

Cooper-Dale, 2018-01-06:
it is %HOMEPATH%\.platformio\penv

ivankravets, 2018-07-26:
Should be fixed with the latest PIP

closed 2018-07-26 by ivankravets