platformio/platformio-atom-ide #1320: PIO Test auto opens a serial monitor at 9600 baudrate

qtphan734, 2017-12-29:
When we run PIO test command, it automatically opens a serial monitor at 9600 baud rate after uploading.  Is there a way to have PIO test open the serial monitor at 115200 baud rate instead?

ivankravets, 2017-12-30:
This value is hardcoded to 9600. I think we can add new option `test_speed` and allow users change it.

Could you explain here why do you need that? We create own Serial instance for each test and use it for test processing.

qtphan734, 2018-01-02:
We are currently using LPUART for the STM32L476RG chip with mbed OS.  This chip requires the system clock to be 32.768 kHz for baud rates between 300 - 9600, which will not work for our project.  We have used the following fix from mbed to get serial to work at 115200: [https://github.com/ARMmbed/mbed-os/issues/5389](https://github.com/ARMmbed/mbed-os/issues/5389).   

However, since PIO test runs at 9600, we cannot see the results, which is needed in our automated test system.  Can you please add that test_speed option?

Thanks!

ivankravets, 2018-01-03:
Moved to https://github.com/platformio/platformio-core/issues/1273

closed 2018-01-03 by ivankravets