platformio/platformio-atom-ide #1321: Uncaught Error: ENOENT: no such file or directory, lstat '~/Documents/PlatformIO/Projects/171229-...

GeoSn0w, 2017-12-29:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.23.1 x64
**Electron**: 1.6.15
**OS**: Mac OS X 10.13.2
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 2.0.0-rc.5

### Stack Trace

Uncaught Error: ENOENT: no such file or directory, lstat '/Users/geosn0w/Documents/PlatformIO/Projects/171229-164837-megaatmega2560/src/micrOS.ino.cpp'

```
At fs.js:902

Error: ENOENT: no such file or directory, lstat '/Users/geosn0w/Documents/PlatformIO/Projects/171229-164837-megaatmega2560/src/micrOS.ino.cpp'
    at fs.lstatSync (fs.js:902:18)
    at Object.fs.lstatSync (ELECTRON_ASAR.js:244:16)
    at Proxy.realpathSync (fs.js:1542:21)
    at Proxy.fs.realpathSync (ELECTRON_ASAR.js:321:29)
    at /packages/platformio-ide/lib/project/helpers.js:32:25
    at getActivePioProject (/packages/platformio-ide/lib/project/helpers.js:36:6)
    at Object.highlightActiveProject (/packages/platformio-ide/lib/maintenance.js:133:13)
    at doHighlight (/packages/platformio-ide/lib/main.js:227:43)
    at Function.module.exports.Emitter.simpleDispatch (/Applications/Atom.app/Contents/Resources/app/node_modules/event-kit/lib/emitter.js:25:20)
    at Emitter.module.exports.Emitter.emit (/Applications/Atom.app/Contents/Resources/app/node_modules/event-kit/lib/emitter.js:141:34)
    at /Applications/Atom.app/Contents/Resources/app/src/workspace.js:550:30
```

### Commands

```
     -2:23.7.0 platformio-ide:home (div.tool-panel.tree-view)
  4x -1:29.4.0 core:backspace (input.hidden-input)
     -1:24.7.0 platformio-ide:target:upload (input.hidden-input)
     -1:24.7.0 platformio-ide:target:upload-/Users/geosn0w/Documents/PlatformIO/Projects/171229-164837-megaatmega2560 (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -1:15 platformio-ide:target:build (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -1:15 platformio-ide:target:build-/Users/geosn0w/Documents/PlatformIO/Projects/171229-164837-megaatmega2560 (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -1:06.2.0 platformio-ide:target:upload (input.hidden-input)
     -1:06.2.0 platformio-ide:target:upload-/Users/geosn0w/Documents/PlatformIO/Projects/171229-164837-megaatmega2560 (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -0:52.2.0 platformio-ide:target:upload (div.tool-panel.tree-view)
     -0:52.2.0 platformio-ide:target:upload-/Users/geosn0w/Documents/PlatformIO/Projects/171229-164837-megaatmega2560 (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -0:45 platformio-ide:target:upload (div.tool-panel.tree-view)
     -0:45 platformio-ide:target:upload-/Users/geosn0w/Documents/PlatformIO/Projects/171229-164837-megaatmega2560 (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
  4x -0:43.7.0 platformio-ide:target:upload (div.tool-panel.tree-view)
     -0:30 platformio-ide:target:upload-/Users/geosn0w/Documents/PlatformIO/Projects/171229-164837-megaatmega2560 (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
  2x -0:18.7.0 build:trigger (atom-workspace.workspace.scrollbars-visible-always.theme-one-dark-syntax.theme-one-dark-ui)
     -0:09.3.0 application:show-settings (div.build.tool-panel.native-key-bindings)
```

### Non-Core Packages

```
autocomplete-clang 0.11.4 
build 0.70.0 
busy-signal 1.4.3 
file-icons 2.1.15 
intentions 1.1.5 
language-ini 1.19.0 
linter 2.2.0 
linter-gcc 0.7.1 
linter-ui-default 1.6.10 
platformio-ide 2.0.0-rc.5 
platformio-ide-debugger 1.2.5 
platformio-ide-terminal 2.7.0 
tool-bar 1.1.0 
```

ivankravets, 2017-12-30:
Could you reproduce this issue after IDE restart?

GeoSn0w, 2017-12-31:
The error appears from time to time. Not always. Sometimes closing the project and reloading it works, sometimes doesn't and I have to restart Atom. There is also a thing with avrdude.It is very flaky. Sometimes it acts like the firmware has been written successfully yet the Arduino board doesn't change anything (I have a TFT attached), sometimes it says "timed out" for no good reason (used to work a few seconds ago).

ivankravets, 2018-01-10:
This is temporary file `ino.cpp`. Please don't touch it.

closed 2018-01-10 by ivankravets