platformio/platformio-atom-ide #1322: C/C++ project index rebuild failed

alekelizalde, 2017-12-29:
EnvironmentError: No tool named 'platformio': not a Zip file:
File "/Users/AlexMAC/.platformio/penv/lib/python2.7/site-packages/platformio/builder/main.py", line 113:
None
File "/Users/AlexMAC/.platformio/packages/tool-scons/script/../engine/SCons/Defaults.py", line 88:
_default_env = SCons.Environment.Environment(*args, **kw)
File "/Users/AlexMAC/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 982:
apply_tools(self, tools, toolpath)
File "/Users/AlexMAC/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 107:
env.Tool(tool)
File "/Users/AlexMAC/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 1788:
tool = SCons.Tool.Tool(tool, toolpath, **kw)
File "/Users/AlexMAC/.platformio/packages/tool-scons/script/../engine/SCons/Tool/__init__.py", line 104:
module = self._tool_module()
File "/Users/AlexMAC/.platformio/packages/tool-scons/script/../engine/SCons/Tool/__init__.py", line 164:
raise SCons.Errors.EnvironmentError(m)
========================= [ERROR] Took 379.53 seconds =========================

Sto-li, 2017-12-30:
I have the same from the update

ivankravets, 2017-12-30:
Please remove `/Users/AlexMAC/.platformio/penv` and restart IDE

closed 2017-12-30 by ivankravets