platformio/platformio-atom-ide #1323: C/C++ project index rebuild failed

mono-baka, 2017-12-30:
Traceback (most recent call last):
  File "C:\Users\fumiya\.platformio\penv\Scripts\platformio-script.py", line 9, in <module>
    load_entry_point('platformio==3.5.1a1', 'console_scripts', 'platformio')()
  File "c:\users\fumiya\.platformio\penv\lib\site-packages\platformio\__main__.py", line 126, in main
    error_str += str(e)
  File "c:\users\fumiya\.platformio\penv\lib\site-packages\platformio\exception.py", line 23, in __str__
    return Exception.__str__(self)
UnicodeEncodeError: 'cp932' codec can't encode character u'\ufffd' in position 263: illegal multibyte sequence

Sto-li, 2017-12-30:
I deleted python form my PC then deleted this folder C:\Users\YOURNAME\.platformio\packages. Now it works.

mono-baka, 2017-12-30:
Oh,this is my mistake!I was strange.
I'm sorry for losing your time.

closed 2017-12-30 by mono-baka