platformio/platformio-atom-ide #1324: C/C++ project index rebuild failed

deandreyanov, 2017-12-30:
Error: Traceback (most recent call last):
  File "c:\users\deand\.platformio\penv\lib\site-packages\platformio\__main__.py", line 120, in main
    cli(None, None, None)
  File "c:\users\deand\.platformio\penv\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\deand\.platformio\penv\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\users\deand\.platformio\penv\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\deand\.platformio\penv\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\deand\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\users\deand\.platformio\penv\lib\site-packages\click\decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\deand\.platformio\penv\lib\site-packages\platformio\commands\init.py", line 102, in cli
    pg = ProjectGenerator(project_dir, ide, env_name)
  File "c:\users\deand\.platformio\penv\lib\site-packages\platformio\ide\projectgenerator.py", line 38, in __init__
    self._gather_tplvars()
  File "c:\users\deand\.platformio\penv\lib\site-packages\platformio\ide\projectgenerator.py", line 155, in _gather_tplvars
    self._tplvars.update(self.get_project_build_data())
  File "c:\users\deand\.platformio\penv\lib\site-packages\platformio\util.py", line 134, in __call__
    value = self.func(*args)
  File "c:\users\deand\.platformio\penv\lib\site-packages\platformio\ide\projectgenerator.py", line 79, in get_project_build_data
    raise result.exception
NotImplementedError

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

ivankravets, 2017-12-30:
Could you provide `platformio.ini` here?

deandreyanov, 2017-12-30:
I updated and reinsalled packeges. And problem go out.

ivankravets, 2017-12-30:
Thanks a lot for the report!

closed 2017-12-30 by ivankravets