platformio/platformio-atom-ide #1348: C/C++ project index rebuild failed

essera, 2018-01-05:
Traceback (most recent call last):
  File "c:\users\andreas\.platformio\penv\lib\site.py", line 703, in <module>
    main()
  File "c:\users\andreas\.platformio\penv\lib\site.py", line 692, in main
    aliasmbcs()
  File "c:\users\andreas\.platformio\penv\lib\site.py", line 515, in aliasmbcs
    import locale, codecs
  File "c:\users\andreas\.platformio\penv\lib\locale.py", line 17, in <module>
    import functools
ImportError: No module named functools

ivankravets, 2018-01-06:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/797

ivankravets, 2018-01-06:
Please uninstall all Pythons from a system using System > Control Panel > Uninstall Pythons.
Restart DE.

closed 2018-01-06 by ivankravets