platformio/platformio-atom-ide #1352: pio home fails missing incremental package

jeffdexter77, 2018-01-07:
Updated atom to 1.23.2, and platformio-ide fails to load PIO Home, with the following error:

```
Traceback (most recent call last):
  File "main.py", line 16, in init main
  File "cmd_debug.py", line 21, in init cmd_debug
  File "/Users/jeffdexter/.platformio/packages/_tmp_installing-z8CNde-package/twisted/__init__.py", line 23, in <module>
    from twisted._version import __version__ as version
  File "/Users/jeffdexter/.platformio/packages/_tmp_installing-z8CNde-package/twisted/_version.py", line 8, in <module>
    from incremental import Version
ImportError: No module named incremental
```

ivankravets, 2018-01-10:
Thanks! Fixed in https://github.com/platformio/platformio-core/commit/ae38d17b7f422c5be24b74f14d3c0fcf92f689e3

Please open PIO IDE Terminal and type `pio upgrade --dev` and restart IDE. All should work now.

closed 2018-01-10 by ivankravets

jeffdexter77, 2018-01-11:
Thanks, confirmed this is fixed.