platformio/platformio-atom-ide #1353: platfromio home page can't load

talkingbeast, 2018-01-07:
hi,everyone.
 i have met this problem seveal times.
 the problem is when i click the platformio home button,the platformio home page don't show.
[image]

  i search for the solution,delete the /home/talkingbeast/.platformio folder works and restart atom works.
[image]

but the problem is very  annoying and this experience is sucks.BTW,this solution is temperary,the same problem will happed after several days.
can someone give me some advice to solve the problem.

ivankravets, 2018-01-10:
What is your OS/distro?

Please provide an output from a system Terminal
```
echo $PATH
echo $PYTHONPATH
python --version
pip --version
pip list
```

talkingbeast, 2018-01-10:
this is the output of my system terminal:

``

ivankravets, 2018-01-10:
Next time when you will have a problem with PIO Home, please open PIO IDE Terminal and type `pio home`. You will some errors. Please share them here. Thanks!

closed 2018-01-10 by ivankravets

talkingbeast, 2018-01-11:
sure thing.thank you. @ivankravets