platformio/platformio-atom-ide #1356: C/C++ project index rebuild failed

vladvana, 2018-01-07:
Error: [01/07/18 17:17:35] Processing nucleo_f103rb (platform: ststm32; board: nucleo_f103rb; framework: mbed)

PackageManager: Installing toolchain-gccarmnoneeabi @ >=1.60301.0
Downloading...
Unpacking...
Verbose mode can be enabled via `-v, --verbose` option
Collected 16 compatible libraries
Scanning dependencies...
Error: Could not find `ualloc` dependency for `mbed-drivers` library
 [ERROR] Took 863.16 seconds

closed 2018-01-07 by vladvana

warlockerino, 2018-03-22:
Did you find a solution ?

ivankravets, 2018-04-25:
Fixed in https://github.com/platformio/platformio-core/commit/2ae169d210cc9c384866720dc6e337405efa8d07

Please open PIO IDE Terminal and type `pio upgrade --dev`.