platformio/platformio-atom-ide #1360: Uncaught Error: Could not start PIO Home server: Traceback (most recent call last):  File "main....

sysopfb, 2018-01-10:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.23.2 x64
**Electron**: 1.6.15
**OS**: Mac OS X 10.11.6
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 2.0.0-rc.5

### Stack Trace

Uncaught Error: Could not start PIO Home server: Traceback (most recent call last):
  File "main.py", line 17, in init main
  File "cmd_home.py", line 10, in init cmd_home
ImportError: No module named configparser

```
At /Users/jasonreaves/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1571

Error: Could not start PIO Home server: Traceback (most recent call last):
  File "main.py", line 17, in init main
  File "cmd_home.py", line 10, in init cmd_home
ImportError: No module named configparser

    at /packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1571:23
    at ChildProcess.onExit (/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:499:5)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at maybeClose (internal/child_process.js:885:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
```

### Commands

```
```

### Non-Core Packages

```
autocomplete-clang 0.11.4 
build 0.70.0 
busy-signal 1.4.3 
file-icons 2.1.15 
intentions 1.1.5 
language-ini 1.19.0 
linter 2.2.0 
linter-gcc 0.7.1 
linter-ui-default 1.6.10 
platformio-ide 2.0.0-rc.5 
platformio-ide-debugger 1.2.5 
platformio-ide-terminal 2.8.0 
tool-bar 1.1.4 
```

ivankravets, 2018-01-10:
Please open PIO IDE Terminal and run `pio update`. Restart Atom. Sorry for the issue. We have just fixed it.

closed 2018-01-10 by ivankravets

Dustify, 2018-01-10:
Confirmed resolved for me, just wanted to say thanks for the quick resolution and keep up the good work!

taedet, 2018-01-17:
I have a similar issue. Installed Atom, and installed package platformio-ide. After restart i get an error message. A dump of the stack:

[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.23.3 x64
**Electron**: 1.6.15
**OS**: Microsoft Windows 7 Enterprise 
**Thrown From**: [platformio-ide](https://github.com/platformio/platformio-atom-ide) package 2.0.0-rc.7

### Stack Trace

Uncaught Error: Could not start PIO Home server: Traceback (most recent call last):
  File "main.py", line 12, in init main
  File "c:\users\terpstrt\.platformio\penv\lib\site-packages\click\__init__.py", line 18, in <module>
    from .core import Context, BaseCommand, Command, MultiCommand, Group, \
  File "c:\users\terpstrt\.platformio\penv\lib\site-packages\click\core.py", line 8, in <module>
    from .types import convert_type, IntRange, BOOL
  File "c:\users\terpstrt\.platformio\penv\lib\site-packages\click\types.py", line 5, in <module>
    from ._compat import open_stream, text_type, filename_to_ui, \
  File "c:\users\terpstrt\.platformio\penv\lib\site-packages\click\_compat.py", line 6, in <module>
    from weakref import WeakKeyDictionary
  File "C:\Users\terpstrt\.platformio\python27\Lib\weakref.py", line 14, in <module>
    from _weakref import (
ImportError: cannot import name _remove_dead_weakref

```
At C:\Users\terpstrt\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1615

Error: Could not start PIO Home server: Traceback (most recent call last):
  File "main.py", line 12, in init main
  File "c:\users\terpstrt\.platformio\penv\lib\site-packages\click\__init__.py", line 18, in <module>
    from .core import Context, BaseCommand, Command, MultiCommand, Group, \
  File "c:\users\terpstrt\.platformio\penv\lib\site-packages\click\core.py", line 8, in <module>
    from .types import convert_type, IntRange, BOOL
  File "c:\users\terpstrt\.platformio\penv\lib\site-packages\click\types.py", line 5, in <module>
    from ._compat import open_stream, text_type, filename_to_ui, \
  File "c:\users\terpstrt\.platformio\penv\lib\site-packages\click\_compat.py", line 6, in <module>
    from weakref import WeakKeyDictionary
  File "C:\Users\terpstrt\.platformio\python27\Lib\weakref.py", line 14, in <module>
    from _weakref import (
ImportError: cannot import name _remove_dead_weakref

    at /packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1615:23
    at ChildProcess.onExit (/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:537:5)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at ChildProcess.cp.emit (/packages/platformio-ide/node_modules/cross-spawn/lib/enoent.js:40:29)
    at maybeClose (internal/child_process.js:885:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
```

### Commands

```
     -2:10.5.0 core:confirm (input.hidden-input)
     -0:57.1.0 platformio-ide:maintenance.open-terminal (atom-pane.pane.active)
```

### Non-Core Packages

```
autocomplete-clang 0.11.4 
build 0.70.0 
busy-signal 1.4.3 
file-icons 2.1.15 
intentions 1.1.5 
language-ini 1.19.0 
linter 2.2.0 
linter-gcc 0.7.1 
linter-ui-default 1.6.10 
platformio-ide 2.0.0-rc.7 
platformio-ide-debugger 1.2.5 
platformio-ide-terminal 2.8.0 
tool-bar 1.1.4 
```

ivankravets, 2018-01-17:
@taedet  https://github.com/platformio/platformio-atom-ide/issues/1358

https://community.platformio.org/t/cant-get-pio-to-work-home-doesnt-want-to-load/2755/15?u=ivankravets

taedet, 2018-01-17:
I already had Python 2.7 installed, and also 3.2 - need that for some other project. The 2.7 version was still installed under my user.platformio folder. Do I have to uninstall 3.2?

ivankravets, 2018-01-17:
Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

Try to remove `%USERPROFILE%\.platformio\penv` folder and restart IDE.

taedet, 2018-01-18:
I removed the penv folder, and restart the IDE. It then proceeded to complete the install for platform-ide and prompted me to restart. Doing that I got te same error.

The vars you requested:

path:
PATH=C:\ProgramData\Oracle\Java\javapath;C:\python27;C:\Python36\Scripts\;C:\Python36\;C:\Program Files\RA2HP\;C:\windows\system32
;C:\windows;C:\windows\System32\Wbem;C:\windows\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Intel\OpenCL SDK\3.0\bin\x
86;C:\Program Files (x86)\Intel\OpenCL SDK\3.0\bin\x64;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\Pro
gram Files\Microsoft SQL Server\110\Tools\Binn\;C:\Program Files (x86)\Microsoft SDKs\TypeScript\1.0\;C:\Program Files\Microsoft S
QL Server\120\Tools\Binn\;C:\windows\System32\WindowsPowerShell\v1.0\;C:\Program Files\Java\jdk1.8.0_101\bin;C:\Program Files\PuTT
Y\;C:\windows\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Common Files\Tom Sawyer Software\8.0.0;C:\Program Files (x86
)\Common Files\Win Wrap\10.25.004;C:\Program Files\LLVM\bin;C:\Users\terpstrt\AppData\Local\atom\bin;C:\ProgramData\terpstrt\atom\
bin;C:\ProgramData\Hewlett-Packard\PC COE\VSC\Drivers\;C:\Users\terpstrt\AppData\Local\atom\bin;C:\ProgramData\terpstrt\atom\bin

PythonPath: c:\python27

python version: Python 2.7

pip --version: pip 9.0.1 from c:\python36\lib\site-packages (python 3.6)

pip list:
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(l
egacy|columns) in your pip.conf under the [list] section) to disable this warning.
beautifulsoup4 (4.5.3)
lxml (3.7.2)
olefile (0.44)
Pillow (4.1.0)
pip (9.0.1)
PyAutoGUI (0.9.35)
PyMsgBox (1.0.6)
PyScreeze (0.1.9)
PyTweening (1.0.3)
pytz (2016.10)
setuptools (32.3.1)
six (1.10.0)
wheel (0.29.0)

ivankravets, 2018-01-18:
@taedet please contact me@ikravets.com for the further details.

taedet, 2018-01-18:
> @taedet<https://github.com/taedet> please contact me@ikravets.com<mailto:me@ikravets.com> for the further details.

Hi Ivan,

can you please give me an idea where to look to solve this? I noticed pip is using the python 3.2 version...

thanks,

Taede

ivankravets, 2018-01-18:
1. System > Control Panel > Programs > Uninstall ALL Pythons
2. Install Python 2.7.14 and add it to PATH: http://docs.platformio.org/en/latest/faq.html#install-python-interpreter
3. Install Python 3
4. Remove `C:\Users\terpstrt\.platformio` folder.

Start IDE.

saouinet, 2018-02-06:
Hi, I still have it. Even after reinstalling 2.7.14 and removing the .platformio 
"c:\users\my.name.platformio\penv\lib\weakref.py", line 14, in from _weakref import ( ImportError: cannot import name _remove_dead_weakref"

ivankravets, 2018-02-06:
@saouinet did you uninstall all Pythons? Removed `c:\users\my.name.platformio`?

ketsa, 2018-02-27:
Same here : 
Uncaught Error: Could not start PIO Home server: Traceback (most recent call last): File "main.py", line 13, in init main File "C:\Users\Matc.platformio\penv\lib\site-packages\platformio\app.py", line 24, in import requests File "C:\Users\Matc.platformio\penv\lib\site-packages\requests_init.py", line 43, in import urllib3 File "C:\Users\Matc.platformio\penv\lib\site-packages\urllib3\init__.py", line 8, in from .connectionpool import ( File "C:\Users\Matc.platformio\penv\lib\site-packages\urllib3\connectionpool.py", line 29, in from .connection import ( File "C:\Users\Matc.platformio\penv\lib\site-packages\urllib3\connection.py", line 39, in from .util.ssl import ( File "C:\Users\Matc.platformio\penv\lib\site-packages\urllib3\util_init__.py", line 6, in from .ssl import ( File "C:\Users\Matc.platformio\penv\lib\site-packages\urllib3\util\ssl_.py", line 4, in import hmac File "C:\Users\Matc.platformio\penv\lib\hmac.py", line 8, in from operator import _compare_digest as compare_digest ImportError: cannot import name _compare_digest
C:\Users\Matc\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1616
Hide Stack Trace
Error: Could not start PIO Home server: Traceback (most recent call last):
  File "main.py", line 13, in init main
  File "C:\Users\Matc\.platformio\penv\lib\site-packages\platformio\app.py", line 24, in <module>
    import requests
  File "C:\Users\Matc\.platformio\penv\lib\site-packages\requests\__init__.py", line 43, in <module>
    import urllib3
  File "C:\Users\Matc\.platformio\penv\lib\site-packages\urllib3\__init__.py", line 8, in <module>
    from .connectionpool import (
  File "C:\Users\Matc\.platformio\penv\lib\site-packages\urllib3\connectionpool.py", line 29, in <module>
    from .connection import (
  File "C:\Users\Matc\.platformio\penv\lib\site-packages\urllib3\connection.py", line 39, in <module>
    from .util.ssl_ import (
  File "C:\Users\Matc\.platformio\penv\lib\site-packages\urllib3\util\__init__.py", line 6, in <module>
    from .ssl_ import (
  File "C:\Users\Matc\.platformio\penv\lib\site-packages\urllib3\util\ssl_.py", line 4, in <module>
    import hmac
  File "C:\Users\Matc\.platformio\penv\lib\hmac.py", line 8, in <module>
    from operator import _compare_digest as compare_digest
ImportError: cannot import name _compare_digest
    at C:\Users\Matc\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1616:23
    at ChildProcess.onExit (C:\Users\Matc\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:538:5)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at ChildProcess.cp.emit (C:\Users\Matc\.atom\packages\platformio-ide\node_modules\cross-spawn\lib\enoent.js:34:29)
    at maybeClose (internal/child_process.js:885:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
The error was thrown from the platformio-ide package. This issue has already been reported.

saouinet, 2018-02-27:
Hi, it's a pro machine I have multiple pys installed (anaconda, 2.4,2.7 ...)
So a simple upgrade of py2.7.14 isn't enough.
Do you have a clue why the import is wrong?

Gyro-Gearless, 2018-03-30:
Have the same problem with a fresh install of PlatformIO - neither "pio update" nor removing that "penv" folder didn't cure the problem, so my getting started with PlatformIO is somewhat stuck...

Here is the stack trace shown in Atom:

Error: Could not start PIO Home server: Traceback (most recent call last):
  File "main.py", line 12, in init main
  File "c:\users\andy\.platformio\penv\lib\site-packages\click\__init__.py", line 18, in <module>
    from .core import Context, BaseCommand, Command, MultiCommand, Group, \
  File "c:\users\andy\.platformio\penv\lib\site-packages\click\core.py", line 8, in <module>
    from .types import convert_type, IntRange, BOOL
  File "c:\users\andy\.platformio\penv\lib\site-packages\click\types.py", line 5, in <module>
    from ._compat import open_stream, text_type, filename_to_ui, \
  File "c:\users\andy\.platformio\penv\lib\site-packages\click\_compat.py", line 6, in <module>
    from weakref import WeakKeyDictionary
  File "C:\Users\andy\.platformio\python27\Lib\weakref.py", line 14, in <module>
    from _weakref import (
ImportError: cannot import name _remove_dead_weakref
    at C:\Users\andy\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1617:23
    at ChildProcess.onExit (C:\Users\andy\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:538:5)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:194:7)
    at ChildProcess.cp.emit (C:\Users\andy\.atom\packages\platformio-ide\node_modules\cross-spawn\lib\enoent.js:34:29)
    at maybeClose (internal/child_process.js:899:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)

Please let me know if i can provide any further details on my setup!

Greetings Gyro

ivankravets, 2018-04-03:
See https://github.com/platformio/platformio-atom-ide/issues/1360#issuecomment-358652429

trimchess, 2018-04-05:
Have the same issue. None of the workarrounds helps. I often have the problem, that PlatformIO works well one day but not the other day. PlatformIO has too much stability problems and also too much updates. I prefere to work with a stable version. After half a year working with PlatformIO i go back to the Programino ide. **This ide has not so nice features like PlatforIO but it works stable.** The idea behind PlatformIO ist very good but it is not possible to work with such an unstable tool. What a bummer!

ivankravets, 2018-04-07:
@trimchess  Sorry for the issue. What is your OS? Could you better explain us about the problems with PIO? It will help to improve it.

Thanks!

trimchess, 2018-04-09:
@ivankravets 
Op-Sys: W10-64Bit-Home Edition
I installed Atam/PlatformIO some month ago. When it works, it es a very good IDE, baut there are too much stability problems.
Issue:
Opening Atom -> PlatformIO -> PlatformIO Home
Uncaught Error: Could not start PIO Home server.
A red window with m
[error_pfio.txt](https://github.com/platformio/platformio-atom-ide/files/1888810/error_pfio.txt)
any, many stacktraces,at the end:
_The error was thrown from the platformio-ide package. **This issue has already been reported**._
I am stopped in workingThe error was thrown from the platformio-ide package. This issue has already

ivankravets, 2018-04-09:
@trimchess could you try to remove `~/.platformio/packages` folder and restart Atom? Does it work now?

trimchess, 2018-04-10:
No, same problem as before. In th attached error file I see:
ImportError: DLL load failed: %1 ist keine zulässige Win32-Anwendung (in English: is not a valid Win32 application.). What mans that?
[error_pfio.txt](https://github.com/platformio/platformio-atom-ide/files/1893321/error_pfio.txt)

markdagraca, 2018-04-11:
@ivankravets im getting the same problem

ivankravets, 2018-04-11:
@trimchess @markdagraca do you have a custom Python installation in a system?

trimchess, 2018-04-12:
@ivankravets Thanks, may be this was the key hint. After uninstallink my other python installations, Atom/PlatformIO works fine again. Now I have one pyton27 installation (c:\Users\Martin\.platformio\python27\).
But for my work, I prefere to work with my one installation. Therefore I will install my own python, independent of Atom/PlatformIO. May be the problem was a 64 Bit installation of python2.7.14(?).
**Remark**
The problem occured "suddenly" on my PC. In my oppinion, Atom/PlatformIO make too much updates in the background.  Always when I start Atom/PlatformIO, I see a lot of pending updates.
**And an other remark**
PlatformIO is a really top IDE to work in the Arduino world. The IDE is fast and it is very nice and easy to work. It has a wonderfull code completion/code highlighting. My single critic is the instability. Therefore keep always in mind to keep the IDE stable.
Thank you for PlatformIO and many thanks for the support!
BR Martin

markdagraca, 2018-04-12:
@ivankravets yes I have anaconda installed  if you want we can teamviewer again

monchi59, 2018-04-12:
@ivankravets I do have the same issue, and I also have a 64bits Python 2.7.14 
do you think this could be the origin of the issue ? I don't know how to act on that though

ivankravets, 2018-04-13:
PlatformIO will use your personal Python installation if it is compatible with Python >2.7.9.

It seems that your previous installation was old or etc. Try the next:

1) Navigate to System Control Panel > Programs
2) Uninstall all Pythons
3) Remove `~/.platformio` folder
4) Install the latest Python 2.7 manually and add it to PATH. See http://docs.platformio.org/en/latest/faq.html#install-python-interpreter 
5) Start PIO IDE.

monchi59, 2018-04-13:
@ivankravets It worked, thanks ! 
I have to say I don't really understand why re-installing the same Python version would change anything but it works ! 
Thank you for the tip and thank you for your work on PIO

trimchess, 2018-04-13:
@ivankravets It works fine now, thanks!

seclorum, 2018-05-11:
Also happens with me on MacOS, and I can only conclude this is happening "because Python", as in: we all have our rules (or not) about how our local Python install is to be used/upgraded/administered (or not), and PlatformIO's attempt to solve this through 'clever shell integration' has actually just made the matters worse.

Solution for me: deinstall PlatformIO IDE, go back to Sublime+make, get on with work ..