platformio/platformio-atom-ide #1364: InstallationManager

vince-br-549, 2018-01-10:
On a Linux system,  it wants clang.exec  I have clang installed but doesn't recognize it.

Error: Can not find Python Interpreter
    at PlatformIOCoreStage._callee8$ (/home/vince/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:2423:23)
    at tryCatch (/home/vince/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:787:40)
    at Generator.invoke [as _invoke] (/home/vince/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1021:22)
    at Generator.prototype.(anonymous function) [as next] (/home/vince/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:839:21)
    at step (/home/vince/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1792:191)
    at /home/vince/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1792:361

ivankravets, 2018-01-10:
Please provide an output from a system Terminal
```
echo $PATH
echo $PYTHONPATH
python --version
pip --version
pip list
```

vince-br-549, 2018-01-10:
vince@desk2 ~]$ echo $PATH
/usr/lib64/qt-3.3/bin:/usr/local/bin:/usr/local/sbin:/usr/bin:/usr/sbin:/bin:/sbin:/home/vince/.local/bin:/home/vince/bin:/home/vince/esp/xtensa-esp32-elf/bin:/home/vince/.local/bin:/home/vince/bin:/home/vince/esp/xtensa-esp32-elf/bin:/home/vince/.local/bin:/home/vince/bin:/home/vince/esp/xtensa-esp32-elf/bin
[vince@desk2 ~]$ echo $PYTHONPATH
/usr/bin/python
[vince@desk2 ~]$ python --version
Python 2.7.5
[vince@desk2 ~]$ pip --version
pip 9.0.1 from /usr/lib/python2.7/site-packages (python 2.7)
[vince@desk2 ~]$ pip list
DEPRECATION: The default format will switch to columns in the future. You
can use --format=(legacy|columns) (or define a format=(legacy|columns) in
your pip.conf under the [list] section) to disable this warning.
arrow (0.12.0)
backports.functools-lru-cache (1.2.1)
backports.ssl-match-hostname (3.4.0.2)
blivet (0.61.15.65)
bottle (0.12.13)
CDDB (1.4)
certifi (2017.11.5)
chardet (3.0.4)
click (5.1)
colorama (0.3.9)
configobj (4.7.2)
coverage (3.6b3)
cupshelpers (1.0)
decorator (3.4.0)
di (0.3)
ethtool (0.8)
firstboot (19.5)
fros (1.0)
html2text (2016.9.19)
idna (2.6)
iniparse (0.4)
initial-setup (0.3.9.40)
IPy (0.75)
javapackages (1.0.0)
kitchen (1.1.1)
langtable (0.0.31)
lockfile (0.12.2)
lxml
(3.2.1)

M2Crypto
(0.21.1)

mutagen
(1.38)

nose
(1.3.7)

ntplib
(0.3.2)

numpy
(1.7.1)

perf
(0.1)

pip
(9.0.1)

platformio
(3.5.1a4)

policycoreutils-default-encoding
(0.1)

pycups
(1.9.63)

pycurl
(7.19.0)

pygobject
(3.22.0)

pygpgme
(0.3)

pyinotify
(0.9.4)

pykickstart
(1.99.66.12)

pyliblzma
(0.5.3)

pyparsing
(1.5.6)

pyparted
(3.9)

pyserial
(3.4)

python-augeas
(0.5.0)

python-dateutil
(1.5)

python-dmidecode
(3.12.2)

python-linux-procfs
(0.4.9)

python-meh
(0.25.2)

python-nss
(0.16.0)

pytoml
(0.1.14)

pytz (2016.10)
pyudev (0.15)
pyxattr (0.5.1)
PyYAML (3.10)
registries (0.1)
requests (2.18.4)
rhsm (1.19.10)
scdate (1.10.6)
schedutils (0.4)
semantic-version (2.6.0)
seobject (0.1)
sepolicy (1.1)
setroubleshoot (1.1)
setuptools (0.9.8)
six (1.9.0)
slip (0.4.0)
slip.dbus (0.4.0)
subscription-manager (1.19.23)
urlgrabber (3.10)
urllib3 (1.22)
wheel (0.30.0)
wxPython (2.8.12.0)
wxPython-common (2.8.12.0)
youtube-dl (2017.12.10)
yum-metadata-parser (1.1.4)
zenmap (6.40)

---
Vince

ivankravets, 2018-01-10:
Minimum supported version for PIO Core + IDE is 2.7.6. Could you update your Python 2?

vince-br-549, 2018-01-10:
whee,   installed  Python 2.7.14

platformio installed but now can't seem to find boards...

vince-br-549, 2018-01-10:
getting error 
PIO Core Call Error: Extra data: line 3 column 1 - line 4 column 63 (char 152956 - 153075)

ivankravets, 2018-01-10:
Could you remove `~/.platformio/penv`folder and restart IDE?

vince-br-549, 2018-01-10:
done.
still complaining:
PIO Core Call Error: Extra data: line 3 column 1 - line 4 column 63 (char
6837 - 6956)

---
Vince

ivankravets, 2018-01-16:
We decreased Python's requirements to 2.7.5+. Could you navigate to Atom > Settings > Packages and uninstall `platformio-ide`. Reopen Atom and install `platformio-ide` again. Does it work?

vince-br-549, 2018-01-16:
I reinstalled and re-started.

Still getting
Could not load boards list
PIO Core Call Error: Extra data: line 3 column 1 - line 4 column 63 (char
152956 - 153075)

---
Vince

ivankravets, 2018-01-16:
Please start Atom and when you will see this error again, open Atom:Menu > View > Developer > Toggle Developer Tools > Console.

Do you see any errors? Please share full log here. Thanks!

vince-br-549, 2018-01-16:
I think this is it.... multiple windows :)

scode: 0data: "Extra data: line 3 column 1 - line 4 column 63 (char 152956
- 153075)"message: "PIO Core Call Error"sagaStack: "at a ↵ JsonRpcError↵
at e (http://127.0.0.1:8010/18cd4320383b16562325.min.js:25:601339)↵    at
Object.w.parse (
http://127.0.0.1:8010/18cd4320383b16562325.min.js:25:603944)↵    at
r.i.onmessage (http://127.0.0.1:8010/18cd4320383b16562325.min.js:11:76113)↵
at r.dispatchEvent (
http://127.0.0.1:8010/18cd4320383b16562325.min.js:25:146599)↵    at
http://127.0.0.1:8010/18cd4320383b16562325.min.js:25:590007↵    at
Array.forEach (native)↵    at r._transportMessage (
http://127.0.0.1:8010/18cd4320383b16562325.min.js:25:589958)↵    at r.emit (
http://127.0.0.1:8010/18cd4320383b16562325.min.js:11:18781)↵    at
WebSocket.ws.onmessage (
http://127.0.0.1:8010/18cd4320383b16562325.min.js:25:575488)"stack:
"JsonRpcError↵    at e (
http://127.0.0.1:8010/18cd4320383b16562325.min.js:25:601339)↵    at
Object.w.parse (
http://127.0.0.1:8010/18cd4320383b16562325.min.js:25:603944)↵    at
r.i.onmessage (http://127.0.0.1:8010/18cd4320383b16562325.min.js:11:76113)↵
at r.dispatchEvent (
http://127.0.0.1:8010/18cd4320383b16562325.min.js:25:146599)↵    at
http://127.0.0.1:8010/18cd4320383b16562325.min.js:25:590007↵    at
Array.forEach (native)↵    at r._transportMessage (
http://127.0.0.1:8010/18cd4320383b16562325.min.js:25:589958)↵    at r.emit (
http://127.0.0.1:8010/18cd4320383b16562325.min.js:11:18781)↵    at
WebSocket.ws.onmessage (
http://127.0.0.1:8010/18cd4320383b16562325.min.js:25:575488)"__proto__:
Error
(anonymous) @ 18cd4320383b16562325.min.js:25
Failed to save to temp variable.

---
Vince

ivankravets, 2018-01-16:
Could you type in PIO IDE Terminal `pio boards --json-output`. Do you see `Extra data: line 3 column 1 - line 4 column 63`?

vince-br-549, 2018-01-17:
in what context?  Obviously a bash window doesn't know PIO from beans.

---
Vince

ivankravets, 2018-01-18:
Please process these commands in PIO IDE Terminal. Also, could you provide TeamViewer access ( me@ikravets.com ) to debug this issue? I would be thankful!

ivankravets, 2018-03-05:
@vince-br-549 could you reproduce this issue?

vince-br-549, 2018-03-06:
I have so many problems I decided to try to "refresh" things.
0. on startup, it notified me of some updates, they happened, then I restarted.
1. rm -Rf ~/.platformio/
2. startup platformio and it re-builds is self.
<code>
sh: module: line 1: syntax error: unexpected end of file
sh: error importing function definition for `BASH_FUNC_module'
Error: You need to specify board ID using `-b` or `--board` option. Supported boards list is available via `platformio boards` command
</code>

tired multiple things.
new project,  fails with boards missing.
Click boards from panel,  hangs.

import arduino fails. waiting for boards.
above get 
<code>
Could not load boards list
PIO Core Call Error: Extra data: line 3 column 1 - line 4 column 63 (char 165322 - 165441)
</code>

ivankravets, 2018-03-06:
Do you use Fedora Linux? It seems like a system's bug https://bugzilla.redhat.com/show_bug.cgi?id=1147043

Nevertheless, I'll work on the fix for Fedora on our side.

vince-br-549, 2018-03-06:
Centos 7.4

------------
Vince

vince-br-549, 2018-03-06:
Team viewer is up

642 552 519

---
Vince

ivankravets, 2018-03-06:
It seems that I don't need TeamViewer. I see the problem with Fedora/Centos distros. I'll deploy new PIO Core+ adn inform you here

ivankravets, 2018-03-07:
Resolved in https://github.com/platformio/platformio-core/issues/1311#issuecomment-371148956

Please open IDE Terminal and type `pio update`. Restart IDE.

Does it work now?

closed 2018-03-07 by ivankravets

vince-br-549, 2018-03-07:
> Resolved in platformio/platformio-core#1311 (comment)

Still getting boards failure..
Could not load boards list
PIO Core Call Error: Extra data: line 3 column 1 - line 4 column 63 (char
167202 - 167321)

---
Vince

ivankravets, 2018-03-07:
Please mail me@ikravets.com with TeamViewer ID/Pass for the session.

vince-br-549, 2018-03-07:
at my normal console
$ python --version
Python 2.7.5

From the cmd line on platformio
[vince@desk2 .atom]$ python --version
Python 2.7.14

I tried updating python using alternatives but this broke yum.
some how platformio is finding the version on /usr/local/bin

Likely
$ set | grep PYTH
PYTHONPATH=/usr/local/bin/python2.7
$

---
Vince

vince-br-549, 2018-03-07:
642 552 519

2264

---
Vince