platformio/platformio-atom-ide #1367: InstallationManager

michael-andreuzza, 2018-01-10:
Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\michael\.platformio\.cache\tmp-3164e5hB16Fr6rLX\virtualenv-14.0.6\virtualenv.py", line 2304, in <module>
    main()
  File "C:\Users\michael\.platformio\.cache\tmp-3164e5hB16Fr6rLX\virtualenv-14.0.6\virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "C:\Users\michael\.platformio\.cache\tmp-3164e5hB16Fr6rLX\virtualenv-14.0.6\virtualenv.py", line 924, in create_environment
    download=download,
  File "C:\Users\michael\.platformio\.cache\tmp-3164e5hB16Fr6rLX\virtualenv-14.0.6\virtualenv.py", line 877, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env)
  File "C:\Users\michael\.platformio\.cache\tmp-3164e5hB16Fr6rLX\virtualenv-14.0.6\virtualenv.py", line 747, in call_subprocess
    "Error %s while executing command %s" % (e, cmd_desc))
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc5 in position 16: ordinal not in range(128)

ivankravets, 2018-01-10:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/1276

ivankravets, 2018-01-10:
Do you use the latest PlatformIO IDE 2.0.0-rc.6?

closed 2018-01-10 by ivankravets

michael-andreuzza, 2018-01-10:
Hej,

Thank you for your help.

/MIKE

⁣Sent from Blue ​

ivankravets, 2018-01-10:
Does it work now?

michael-andreuzza, 2018-01-10:
It has actually worked, I closed it then opened and it worked. Sorry for bothering with such a question, I am brand new in web dev.

/mike

⁣Sent from Blue ​