platformio/platformio-atom-ide #1368: InstallationManager

micrater, 2018-01-10:
Error: Can not find Python Interpreter
    at PlatformIOCoreStage._callee8$ (/home/micra/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:2464:23)
    at tryCatch (/home/micra/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:828:40)
    at GeneratorFunctionPrototype.invoke [as _invoke] (/home/micra/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1062:22)
    at GeneratorFunctionPrototype.prototype.(anonymous function) [as next] (/home/micra/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:880:21)
    at step (/home/micra/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1833:191)
    at /home/micra/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:1833:361

ivankravets, 2018-01-10:
Minimum supported Python version is 2.7.6. Please check `python --version` and upgrade Python 2.

ivankravets, 2018-01-10:
What is your Python version?

micrater, 2018-01-11:
exact message PlatformIO: Can not find Python 2.7 Interpreter
PlatformIO Core is written in Python and depends on it. 
Please install Python 2.7 (PYTHON 3 IS NOT SUPPORTED YET) or if you have it, 
please choose a directory where "python/python.exe" program is located.

echo $PATH
/usr/local/bin:/usr/local/sbin:/usr/bin:/usr/sbin:/bin:/sbin:/home/micra/.local/bin:/home/micra/bin
OS centos 7
python 2.75

micrater, 2018-01-11:
why python.exe !?

ivankravets, 2018-01-11:
Currently, PIO IDE requires Python 2.7.6+. I don't remember, but someone reported that it does work with <2.7.6. We can try again.

Please perform the next commands:

```
sudo pip install virtualenv
cd /tmp
virtualenv piotest
piotest\bin\activate
(piotest) pip install platformio
(piotest) platformio home
```

Does it work?

micrater, 2018-01-11:
with python 2.7.5
it work, ok for platformio but without atom editor 
and now ...

ivankravets, 2018-01-11:
Please try to UNINSTALL our extension and INSTALL again. It should install a new version of our installer where Python 2.7.5 is allowed. Does it work now?

micrater, 2018-01-12:
UNINSTALL and INSTALL platformio via atom interface
==> it is OK with :
Platformio Core 3.5.1a5
cat /etc/centos-release
CentOS Linux release 7.4.1708 (Core)
python
Python 2.7.5 (default, Aug  4 2017, 00:39:18) 
[GCC 4.8.5 20150623 (Red Hat 4.8.5-16)] on linux2
thanks

ivankravets, 2018-01-12:
Does it work now including PIO Home?

micrater, 2018-01-12:
Yes it work now including PIO Home.
Home 0.6.1·Core 3.5.1a5
Happy new year