platformio/platformio-atom-ide #1372: InstallationManager

markdagraca, 2018-01-11:
PIP: The system cannot find the path specified.
Error: spawn C:\Users\Mark D\.platformio\penv\Scripts\pip ENOENT

Tried deleting .platformio and .atom and it still wont work

ivankravets, 2018-01-11:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/1243

ivankravets, 2018-01-11:
Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

closed 2018-01-11 by ivankravets

markdagraca, 2018-01-11:
C:\Users\Mark D>echo %PATH%
C:\Program Files (x86)\Intel\iCLS Client\;C:\Program Files\Intel\iCLS Client\;C:\ProgramData\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\PuTTY\;C:\Program Files (x86)\Skype\Phone\;C:\Program Files\MySQL\MySQL Utilities 1.6\;C:\Program Files (x86)\QuickTime\QTSystem\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files\dotnet\;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\Git\cmd;C:\Python27;C:\Python27\Scripts;C:\Users\Mark D\Anaconda3;C:\Users\Mark D\Anaconda3\Library\mingw-w64\bin;C:\Users\Mark D\Anaconda3\Library\usr\bin;C:\Users\Mark D\Anaconda3\Library\bin;C:\Users\Mark D\Anaconda3\Scripts;C:\Users\Mark D\AppData\Local\Microsoft\WindowsApps;C:\Program Files (x86)\Microsoft VS Code\bin;C:\Users\Mark D\AppData\Roaming\Dashlane\4.8.7.35982\bin\Firefox_Extension\{442718d9-475e-452a-b3e1-fb1ee16b8e9f}\components;C:\Users\Mark D\AppData\Roaming\Dashlane\4.8.7.35982\ucrt;C:\Users\Mark D\AppData\Local\atom\bin;C:\Users\Mark D\AppData\Local\Microsoft\WindowsApps;C:\Program Files\CMake\bin;C:\Users\Mark D\AppData\Roaming\Dashlane\5.3.2.14186\bin\Firefox_Extension\{442718d9-475e-452a-b3e1-fb1ee16b8e9f}\components;C:\Users\Mark D\AppData\Roaming\Dashlane\5.3.2.14186\ucrt;C:\Program Files\LLVM\bin

C:\Users\Mark D>echo %PYTHONPATH%
%PYTHONPATH%

C:\Users\Mark D>python --version
Python 2.7.14

C:\Users\Mark D>pip --version
pip 9.0.1 from c:\python27\lib\site-packages (python 2.7)

C:\Users\Mark D>pip list
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
arrow (0.12.0)
backports.functools-lru-cache (1.2.1)
bottle (0.12.13)
certifi (2017.11.5)
chardet (3.0.4)
click (5.1)
colorama (0.3.9)
geolocation-python (0.2.2)
idna (2.6)
lockfile (0.12.2)
numpy (1.10.2)
pip (9.0.1)
platformio (3.5.0)
pyserial (3.4)
python-dateutil (2.6.1)
requests (2.18.4)
semantic-version (2.6.0)
setuptools (28.8.0)
six (1.11.0)
urllib3 (1.22)

ivankravets, 2018-01-11:
Could you try to remove `C:\Users\Mark D\.platformio\penv` folder and restart?

markdagraca, 2018-01-11:
I dont have that folder

markdagraca, 2018-01-11:
Directory: C:\Users\Mark D\.platformio

Mode                LastWriteTime         Length Name
----                -------------         ------ ----
d-----        1/11/2018  12:58 PM                .cache
d-----       12/22/2017  10:11 PM                lib
d-----        1/10/2018   2:59 PM                packages
d-----       11/22/2017   6:18 PM                platforms
-a----        1/10/2018   2:59 PM            215 appstate.json
-a----        1/10/2018   2:59 PM            702 homestate.json

ivankravets, 2018-01-11:
Could you provide https://www.teamviewer.com/en/ session to debug this issue? Please contact me@ikravets.com. I would be thankful!

markdagraca, 2018-01-11:
`conda.exe update -n base conda` helped!