platformio/platformio-atom-ide #1374: InstallationManager

wumfi, 2018-01-11:
Virtualenv Create: Traceback (most recent call last):
  File "/Users/pcundall/.platformio/.cache/tmp-1034nEI96cT9DrnF/virtualenv-14.0.6/virtualenv.py", line 2304, in <module>
    main()
  File "/Users/pcundall/.platformio/.cache/tmp-1034nEI96cT9DrnF/virtualenv-14.0.6/virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "/Users/pcundall/.platformio/.cache/tmp-1034nEI96cT9DrnF/virtualenv-14.0.6/virtualenv.py", line 924, in create_environment
    download=download,
  File "/Users/pcundall/.platformio/.cache/tmp-1034nEI96cT9DrnF/virtualenv-14.0.6/virtualenv.py", line 843, in install_wheel
    cert_data = pkgutil.get_data("pip._vendor.requests", "cacert.pem")
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/pkgutil.py", line 578, in get_data
    loader = get_loader(package)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/pkgutil.py", line 464, in get_loader
    return find_loader(fullname)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/pkgutil.py", line 474, in find_loader
    for importer in iter_importers(fullname):
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/pkgutil.py", line 430, in iter_importers
    __import__(pkg)
  File "/Users/pcundall/.platformio/.cache/tmp-1034nEI96cT9DrnF/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/__init__.py", line 13, in <module>
  File "/Users/pcundall/.platformio/.cache/tmp-1034nEI96cT9DrnF/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/utils/__init__.py", line 26, in <module>
  File "/Users/pcundall/.platformio/.cache/tmp-1034nEI96cT9DrnF/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 3130, in <module>
  File "/Users/pcundall/.platformio/.cache/tmp-1034nEI96cT9DrnF/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 3116, in _call_aside
  File "/Users/pcundall/.platformio/.cache/tmp-1034nEI96cT9DrnF/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 3155, in _initialize_master_working_set
  File "/Users/pcundall/.platformio/.cache/tmp-1034nEI96cT9DrnF/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 956, in subscribe
  File "/Users/pcundall/.platformio/.cache/tmp-1034nEI96cT9DrnF/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 3155, in <lambda>
  File "/Users/pcundall/.platformio/.cache/tmp-1034nEI96cT9DrnF/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 2650, in activate
  File "/Users/pcundall/.platformio/.cache/tmp-1034nEI96cT9DrnF/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 2744, in insert_on
  File "/Users/pcundall/.platformio/.cache/tmp-1034nEI96cT9DrnF/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 2775, in check_version_conflict
  File "/Users/pcundall/.platformio/.cache/tmp-1034nEI96cT9DrnF/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 2643, in _get_metadata
  File "/Users/pcundall/.platformio/.cache/tmp-1034nEI96cT9DrnF/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 1624, in get_metadata_lines
  File "/Users/pcundall/.platformio/.cache/tmp-1034nEI96cT9DrnF/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 1616, in get_metadata
  File "/Users/pcundall/.platformio/.cache/tmp-1034nEI96cT9DrnF/virtualenv-14.0.6/virtualenv_support/pip-8.0.2-py2.py3-none-any.whl/pip/_vendor/pkg_resources/__init__.py", line 1727, in _get
IOError: [Errno 13] Permission denied: '/Library/Python/2.7/site-packages/protobuf-3.2.0-py2.7.egg/EGG-INFO/namespace_packages.txt'

ivankravets, 2018-01-11:
Please provide an output from a system Terminal
```
echo $PATH
echo $PYTHONPATH
python --version
pip --version
pip list
```

wumfi, 2018-01-12:
Hi,

I resolved this in the end by chmod'ing the file listed in the "Permission denied" section to 777.  After each permission change, I got further identical errors for different files.  After going through a chmod/install/chmod/install loop several times, I was able to complete the installation.

A poor solution I'm sure you'll agree, but this was only on a temporary VM and I needed to get it working quickly.

For completeness, (and hopefully a better way of solving this!) your requested output below:

echo $PATH
/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Library/Frameworks/Mono.framework/Versions/Current/Commands:/Applications/Wireshark.app/Contents/MacOS

echo $PYTHONPATH <-- Returned NULL

python --version
Python 2.7.10

pip --version
pip 9.0.1 from /Library/Python/2.7/site-packages (python 2.7)

pip list
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
altgraph (0.10.2)
ansible (2.2.0.0)
bdist-mpkg (0.5.0)
bonjour-py (0.3)
bottle (0.12.13)
certifi (2017.11.5)
cffi (1.9.1)
chardet (3.0.4)
click (5.1)
colorama (0.3.9)
cryptography (1.6)
enum34 (1.1.6)
idna (2.6)
ipaddress (1.0.17)
Jinja2 (2.8)
lockfile (0.12.2)
macholib (1.5.1)
MarkupSafe (0.23)
matplotlib (1.3.1)
modulegraph (0.10.4)
numpy (1.8.0rc1)
paramiko (2.0.2)
pip (9.0.1)
protobuf (3.2.0)
py2app (0.7.3)
pyasn1 (0.1.9)
pycparser (2.17)
pycrypto (2.6.1)
pyobjc-core (2.5.1)
pyobjc-framework-Accounts (2.5.1)
pyobjc-framework-AddressBook (2.5.1)
pyobjc-framework-AppleScriptKit (2.5.1)
pyobjc-framework-AppleScriptObjC (2.5.1)
pyobjc-framework-Automator (2.5.1)
pyobjc-framework-CFNetwork (2.5.1)
pyobjc-framework-Cocoa (2.5.1)
pyobjc-framework-Collaboration (2.5.1)
pyobjc-framework-CoreData (2.5.1)
pyobjc-framework-CoreLocation (2.5.1)
pyobjc-framework-CoreText (2.5.1)
pyobjc-framework-DictionaryServices (2.5.1)
pyobjc-framework-EventKit (2.5.1)
pyobjc-framework-ExceptionHandling (2.5.1)
pyobjc-framework-FSEvents (2.5.1)
pyobjc-framework-InputMethodKit (2.5.1)
pyobjc-framework-InstallerPlugins (2.5.1)
pyobjc-framework-InstantMessage (2.5.1)
pyobjc-framework-LatentSemanticMapping (2.5.1)
pyobjc-framework-LaunchServices (2.5.1)
pyobjc-framework-Message (2.5.1)
pyobjc-framework-OpenDirectory (2.5.1)
pyobjc-framework-PreferencePanes (2.5.1)
pyobjc-framework-PubSub (2.5.1)
pyobjc-framework-QTKit (2.5.1)
pyobjc-framework-Quartz (2.5.1)
pyobjc-framework-ScreenSaver (2.5.1)
pyobjc-framework-ScriptingBridge (2.5.1)
pyobjc-framework-SearchKit (2.5.1)
pyobjc-framework-ServiceManagement (2.5.1)
pyobjc-framework-Social (2.5.1)
pyobjc-framework-SyncServices (2.5.1)
pyobjc-framework-SystemConfiguration (2.5.1)
pyobjc-framework-WebKit (2.5.1)
pyOpenSSL (0.13.1)
pyparsing (2.0.1)
pyserial (3.4)
python-axolotl (0.1.35)
python-axolotl-curve25519 (0.1)
python-dateutil (1.5)
pytz (2013.7)
PyYAML (3.12)
requests (2.18.4)
scipy (0.13.0b1)
semantic-version (2.6.0)
setuptools (18.5)
six (1.4.1)
urllib3 (1.22)
vboxapi (1.0)
wheel (0.29.0)
xattr (0.6.4)
yowsup2 (2.5.0)
zope.interface (4.1.1)

wumfi, 2018-01-12:
As a further update, this error also presents itself in VSCode as well as Atom, so I can say that it probably isn't IDE environment dependent.

ivankravets, 2018-06-20:
Should be fixed in the latest release. Please upgrade.

ivankravets, 2018-06-21:
Should be fixed in the latest release. Please update `platformio-ide` package via `Settings > Updates`.

closed 2018-06-21 by ivankravets