platformio/platformio-atom-ide #1386: InstallationManager

EttNollEtt, 2018-01-13:
Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\Kalle\.platformio\.cache\tmp-7524fTgGpsiZ2UdH\virtualenv-14.0.6\virtualenv.py", line 2304, in <module>
    main()
  File "C:\Users\Kalle\.platformio\.cache\tmp-7524fTgGpsiZ2UdH\virtualenv-14.0.6\virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "C:\Users\Kalle\.platformio\.cache\tmp-7524fTgGpsiZ2UdH\virtualenv-14.0.6\virtualenv.py", line 924, in create_environment
    download=download,
  File "C:\Users\Kalle\.platformio\.cache\tmp-7524fTgGpsiZ2UdH\virtualenv-14.0.6\virtualenv.py", line 834, in install_wheel
    from urllib.parse import urljoin
  File "C:\Users\Kalle\.platformio\python27\lib\urllib.py", line 26, in <module>
    import socket
  File "C:\Users\Kalle\.platformio\python27\lib\socket.py", line 47, in <module>
    import _socket
ImportError: Module use of python36.dll conflicts with this version of Python.

ivankravets, 2018-01-16:
Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

EttNollEtt, 2018-01-16:
C:\ProgramData\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files (x86)\Skype\Phone\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files\MATLAB\R2017b\runtime\win64;C:\Program Files\MATLAB\R2017b\bin;C:\Program Files\LLVM\bin;C:\Users\Kalle\AppData\Local\Microsoft\WindowsApps;C:\Users\Kalle\AppData\Local\Programs\Python\Python36-32;C:\Users\Kalle\AppData\Local\Microsoft\WindowsApps;C:\Users\Kalle\AppData\Local\GitHubDesktop\bin;C:\Users\Kalle\AppData\Local\atom\bin

C:\Users\Kalle\AppData\Local\Programs\Python\Python36-32\DLLs;C:\Users\Kalle\Desktop

Python 3.6.3

And:
'pip' is not recognized as an internal or external command,
operable program or batch file.

ivankravets, 2018-01-16:
Did you install Python 3 manually?

EttNollEtt, 2018-01-17:
No, and this shows up when I start atom and I think it started when I tried to make it work for MATLAB.

ivankravets, 2018-01-17:
1. Ok, please navigate to System > Control Panel > Programs and uninstall all Pythons from a system.
2. Please remove `C:\Users\Kalle\.platformio` folder and restart IDE.

ivankravets, 2018-06-19:
Please reopen if you still need help

closed 2018-06-19 by ivankravets

catafest, 2019-05-23:
do you enable antivirus?!