platformio/platformio-atom-ide #1390: C/C++ project index rebuild failed

ellingsson, 2018-01-14:
Error: [Sun Jan 14 22:15:03 2018] Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Collected 25 compatible libraries
Scanning dependencies...
Error: Could not find `ESPAsyncWebServer` dependency for `Basecamp` library
========================== [ERROR] Took 0.86 seconds ==========================

ellingsson, 2018-01-14:
Thanks for a great IDE. It seems that the dependency states "ESPAsyncWebServer" while the package name is "ESP Async Webserver". Is this the case?

ivankravets, 2018-01-16:
Fixed in https://github.com/merlinschumacher/Basecamp/pull/13

Please wait when PR will be merged and PIO update this library.

closed 2018-01-16 by ivankravets