platformio/platformio-atom-ide #1393: C/C++ project index rebuild failed

westward, 2018-01-15:
Error: Could not install 'framework-arduinoespressif8266' with version requirements '~1.20400.0' for your system 'windows_amd64'.
More details: http://bit.ly/faq-package-manager

ivankravets, 2018-01-16:
Have you read this http://bit.ly/faq-package-manager ?

Did you try all solutions?

ivankravets, 2018-01-16:
Please reopen if you still need help

closed 2018-01-16 by ivankravets

westward, 2018-01-16:
Ivan,
The IDE offered me a dialog with a button like Submit report and I confirmed it. No more and no less.

ivankravets, 2018-01-16:
Does it work now?

westward, 2018-01-16:
Probably yes.
The trouble happens on a PlatformIO Home page. The IDE displayed a red bullet notification about two available Platforms updates: AtmelAVR and Espressif 8266. On a update of second one the error happens. When IDE was restarted, the update notification and error disappears. Don't know has it auto fixed or just hidden. As I may see, now IDE works.