platformio/platformio-atom-ide #1395: PlatformIO Home: "Show at startup" checkbox not working

wurmtal868, 2018-01-16:
### Configuration

**Operating system**: Windows 7

**PlatformIO Version** (`platformio --version`): Home 0.6.1·Core 3.5.1a6

### Description of problem
"Show at startup" has no effect. PlatformIO Home will be shown no matter if the box is checked or not

#### Steps to Reproduce

1. Uncheck "Show at startup" on Home tab
2. Close the HOME tab
3. Restart Atom.IO 

### Actual Results

Atom.IO will open PlatformIO Home tab in foreground

### Expected Results

Atom.IO should restore the last state of tabs without the PlatformIO Home tab

ivankravets, 2018-01-16:
Did you overwrite HOME environment variable?

wurmtal868, 2018-01-17:
I have a HOME variable defined in Win 7 and PlatformIO is using this folder. I have not defined any HOME in PlatformIO-IDE

ivankravets, 2018-01-17:
It seems that JS does not handle it in the same way as Python does. Please add extra http://docs.platformio.org/en/latest/envvars.html#envvar-PLATFORMIO_HOME_DIR environment variable.

wurmtal868, 2018-01-17:
The extra HOME variable did the trick. 
This is nothing I would do intuitive.

ivankravets, 2018-01-17:
Ok, let's debug it together. 

https://github.com/atom/fs-plus/blob/master/src/fs-plus.coffee#L20

Please provide an output of

```
echo %HOME%
echo %USERPROFILE%
```

wurmtal868, 2018-01-17:
Here is the output:

C:\Users\wurmtal>echo %HOME%
c:\home

C:\Users\wurmtal>echo %USERPROFILE%
C:\Users\wurmtal

ivankravets, 2018-01-17:
Good, I made PR https://github.com/atom/fs-plus/pull/44

If they don't merge it, I'll implement own `getHomeDirectory` for Node.JS.

ivankravets, 2018-02-17:
Should work in the latest release

closed 2018-02-17 by ivankravets

wurmtal868, 2018-02-26:
I removed the environment variable.
It's working fine.
Thanks!