platformio/platformio-atom-ide #1396: InstallationManager

Wernerhi, 2018-01-16:
Virtualenv Create: Traceback (most recent call last):
  File "/Users/olaf 1/.platformio/.cache/tmp-15183ZkIb26ZA2ogx/virtualenv-14.0.6/virtualenv.py", line 2304, in <module>
    main()
  File "/Users/olaf 1/.platformio/.cache/tmp-15183ZkIb26ZA2ogx/virtualenv-14.0.6/virtualenv.py", line 705, in main
    symlink=options.symlink)
  File "/Users/olaf 1/.platformio/.cache/tmp-15183ZkIb26ZA2ogx/virtualenv-14.0.6/virtualenv.py", line 924, in create_environment
    download=download,
  File "/Users/olaf 1/.platformio/.cache/tmp-15183ZkIb26ZA2ogx/virtualenv-14.0.6/virtualenv.py", line 877, in install_wheel
    call_subprocess(cmd, show_stdout=False, extra_env=env)
  File "/Users/olaf 1/.platformio/.cache/tmp-15183ZkIb26ZA2ogx/virtualenv-14.0.6/virtualenv.py", line 783, in call_subprocess
    % (cmd_desc, proc.returncode))
OSError: Command "/Users/olaf 1/.platformio/penv/bin/python" -c "import sys, pip; sys...d\"] + sys.argv[1:]))" setuptools pip wheel failed with error code 2

ivankravets, 2018-01-16:
Please provide an output from a system Terminal
```
echo $PATH
echo $PYTHONPATH
python --version
pip --version
pip list
```

This is a known issue. Could you provide TeamViewer session to debug it? I think other users would be thankful if we find a solution. You can contact me@ikravets.com. Thanks!

ivankravets, 2018-01-16:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/1397#issuecomment-358073768

closed 2018-01-16 by ivankravets