platformio/platformio-atom-ide #1397: InstallationManager

Wernerhi, 2018-01-16:
PIP: Error: spawn /Users/olaf 1/.platformio/penv/bin/pip ENOENT

ivankravets, 2018-01-16:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/1243

Wernerhi, 2018-01-16:
echo $PYTHONPATH
python --version
pip --version
pip list
[Blasting-MacBook-Pro-2:~] olaf% echo $PYTHONPATH
PYTHONPATH: Undefined variable.
[Blasting-MacBook-Pro-2:~] olaf% python --version
Python 2.7.9
[Blasting-MacBook-Pro-2:~] olaf% pip --version
pip 1.5.6 from /Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages (python 2.7)
[Blasting-MacBook-Pro-2:~] olaf% pip list
ecdsa (0.13)
esptool (2.2)
pip (1.5.6)
pyaes (1.6.1)
pyserial (3.4)
setuptools (7.0)
wsgiref (0.1.2)
[Blasting-MacBook-Pro-2:~] olaf% 

> Am 16.01.2018 um 19:34 schrieb Ivan Kravets <notifications@github.com>:
> 
> echo $PATH
> echo $PYTHONPATH
> python --version
> pip --version
> pip list
>

ivankravets, 2018-01-16:
Please start Atom and when you will see this error again, open Atom:Menu > View > Developer > Toggle Developer Tools > Console.

Do you see any errors? Please share full log here. Thanks!

Wernerhi, 2018-01-16:
Hi Ivan
Team Viewer ID:513 855 044
Password:5648

/Users/olaf 1/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:480 runCommand platformio Array[1] Object
manager.js [sm]:82 Error: spawn platformio ENOENT
(anonymous) @ manager.js [sm]:82
/Users/olaf 1/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:480 runCommand conda Array[1] Object
/Users/olaf 1/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:480 runCommand /Users/olaf 1/.platformio/penv/bin/python Array[2] Object
/Users/olaf 1/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:480 runCommand python2.7 Array[2] Object
/Users/olaf 1/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:480 runCommand virtualenv Array[3] Object
/Users/olaf 1/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:2486 User's Virtualenv: Error: spawn virtualenv ENOENT
_callee8$ @ /Users/olaf 1/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:2486
/Users/olaf 1/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:480 runCommand python2.7 Array[2] Object
/Users/olaf 1/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:480 runCommand /Users/olaf 1/.platformio/penv/bin/pip Array[4] Object
/Users/olaf 1/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:2223 PIP: Error: spawn /Users/olaf 1/.platformio/penv/bin/pip ENOENT
_callee5$ @ /Users/olaf 1/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:2223
/Users/olaf 1/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:480 runCommand /Users/olaf 1/.platformio/penv/bin/pip Array[3] Object
utils.js [sm]:33 InstallationManager PIP: Error: spawn /Users/olaf 1/.platformio/penv/bin/pip ENOENT
notifyError @ utils.js [sm]:33
internal/process/next_tick.js:103 Uncaught (in promise) undefined

ivankravets, 2018-01-16:
Please don't share TeamViewer settings in public thread. Contact me directly via me@ikravets.com

Thanks!

Wernerhi, 2018-02-15:
<embedded>:121025 IntersectionObserver.observe(target): target element is not a descendant of root.
TreeView @ <embedded>:121025
/Users/olaf 1/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:481 runCommand platformio Array[1] Object
manager.js [sm]:82 Error: spawn platformio ENOENT
(anonymous) @ manager.js [sm]:82
/Users/olaf 1/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:481 runCommand conda Array[1] Object
/Users/olaf 1/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:481 runCommand /Users/olaf 1/.platformio/penv/bin/python Array[2] Object
/Users/olaf 1/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:481 runCommand python2.7 Array[2] Object
/Users/olaf 1/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:481 runCommand virtualenv Array[3] Object
/Users/olaf 1/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:2493 User's Virtualenv: Error: spawn virtualenv ENOENT
_callee8$ @ /Users/olaf 1/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:2493
/Users/olaf 1/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:481 runCommand python2.7 Array[2] Object
/Users/olaf 1/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:481 runCommand /Users/olaf 1/.platformio/penv/bin/pip Array[4] Object
/Users/olaf 1/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:2226 PIP: Error: spawn /Users/olaf 1/.platformio/penv/bin/pip ENOENT
_callee5$ @ /Users/olaf 1/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:2226
/Users/olaf 1/.atom/packages/platformio-ide/node_modules/platformio-node-helpers/lib/index.js:481 runCommand /Users/olaf 1/.platformio/penv/bin/pip Array[3] Object
utils.js [sm]:33 InstallationManager PIP: Error: spawn /Users/olaf 1/.platformio/penv/bin/pip ENOENT
notifyError @ utils.js [sm]:33
internal/process/next_tick.js:103 Uncaught (in promise) undefined
<embedded>:75145 Uncaught (in promise) Error: The workspace can only contain one instance of item [object Object]
    at ItemRegistry.module.exports.ItemRegistry.addItem (/Applications/Atom.app/Contents/Resources/app/src/item-registry.js:11:21)
    at PaneContainer.didAddPaneItem (/Applications/Atom.app/Contents/Resources/app/src/pane-container.js:269:29)
    at Pane.addItem (/Applications/Atom.app/Contents/Resources/app/src/pane.js:632:48)
    at Pane.activateItem (/Applications/Atom.app/Contents/Resources/app/src/pane.js:574:18)
    at Workspace.<anonymous> (/Applications/Atom.app/Contents/Resources/app/src/workspace.js:1175:20)
    at Generator.next (<anonymous>)
    at step (/Applications/Atom.app/Contents/Resources/app/src/workspace.js:11:279)

closed 2018-03-05 by ivankravets