platformio/platformio-atom-ide #1401: C/C++ project index rebuild failed

robinsonp73, 2018-01-17:
I have looked through past references and tried all the mentioned fixes I could find, but still stuck with this problem. Any help appreciated.

echo %PATH%
%PATH%
echo %PYTHONPATH%
%PYTHONPATH%
Python 2.7.14
pip 9.0.1 from c:\users\pedro.fusion\.platformio\penv\lib\site-packages (python 2.7)

pip list
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.

bottle (0.12.13)
certifi (2017.11.5)
chardet (3.0.4)
click (5.1)
colorama (0.3.9)
idna (2.6)
lockfile (0.12.2)
pip (9.0.1)
platformio (3.5.1a7)
pyserial (3.4)
requests (2.18.4)
semantic-version (2.6.0)
setuptools (38.4.0)
urllib3 (1.22)
wheel (0.30.0)

Error: Traceback (most recent call last):
  File "c:\users\pedro.fusion\.platformio\penv\lib\site-packages\platformio\__main__.py", line 120, in main
    cli(None, None, None)
  File "c:\users\pedro.fusion\.platformio\penv\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\pedro.fusion\.platformio\penv\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\users\pedro.fusion\.platformio\penv\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\pedro.fusion\.platformio\penv\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\pedro.fusion\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\users\pedro.fusion\.platformio\penv\lib\site-packages\click\decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\pedro.fusion\.platformio\penv\lib\site-packages\platformio\commands\init.py", line 103, in cli
    pg.generate()
  File "c:\users\pedro.fusion\.platformio\penv\lib\site-packages\platformio\ide\projectgenerator.py", line 124, in generate
    self._render_tpl(tpl_path).encode("utf8"))
  File "c:\users\pedro.fusion\.platformio\penv\lib\site-packages\platformio\ide\projectgenerator.py", line 130, in _render_tpl
    return bottle.template(content, **self._tplvars)
  File "C:\Users\Pedro.Fusion\.platformio\penv\Scripts\bottle.py", line 3619, in template
    return TEMPLATES[tplid].render(kwargs)
  File "C:\Users\Pedro.Fusion\.platformio\penv\Scripts\bottle.py", line 3409, in render
    self.execute(stdout, env)
  File "C:\Users\Pedro.Fusion\.platformio\penv\Scripts\bottle.py", line 3396, in execute
    eval(self.co, env)
  File "<string>", line 2, in <module>
AttributeError: 'NoneType' object has no attribute 'replace'

ivankravets, 2018-01-27:
Please provide a contents of `platformio.ini` file

robinsonp73, 2018-01-28:
[platformio]
env_default = node
src_dir = src
data_dir = src/data

[common]
platform = https://github.com/platformio/platform-espressif8266.git#v1.5.0
build_flags = -g -DMQTT_MAX_PACKET_SIZE=400
debug_flags = -DDEBUG_ESP_CORE -DDEBUG_ESP_SSL -DDEBUG_ESP_WIFI -DDEBUG_ESP_HTTP_CLIENT -DDEBUG_ESP_HTTP_UPDATE -DDEBUG_ESP_HTTP_SERVER -DDEBUG_ESP_UPDATER -DDEBUG_ESP_OTA -DDEBUG_TLS_MEM
lib_deps =
    https://github.com/xoseperez/Time
    ArduinoJson
    https://github.com/me-no-dev/ESPAsyncTCP#a57560d
    https://github.com/me-no-dev/ESPAsyncWebServer#313f337
    https://github.com/marvinroger/async-mqtt-client#v0.8.1
    PubSubClient
    Embedis
    NtpClientLib
	RFM69@7f140f04da
	https://github.com/xoseperez/SPIFlash.git
    https://bitbucket.org/xoseperez/justwifi.git#1.1.6
    https://bitbucket.org/xoseperez/nofuss.git#0.2.5
lib_ignore = RFM69_OTA

[env:test]
platform = ${common.platform}
framework = arduino
board = d1_mini
lib_deps = ${common.lib_deps}
lib_ignore = ${common.lib_ignore}
build_flags = ${common.build_flags} -DNOWSAUTH
upload_speed = 460800
monitor_baud = 115200

[env:wire]
platform = ${common.platform}
framework = arduino
board = esp12e
lib_deps = ${common.lib_deps}
lib_ignore = ${common.lib_ignore}
build_flags = ${common.build_flags}
upload_speed = 460800
monitor_baud = 115200

[env:ota]
platform = ${common.platform}
framework = arduino
board = esp12e
lib_deps = ${common.lib_deps}
lib_ignore = ${common.lib_ignore}
build_flags = ${common.build_flags}
upload_speed = 115200
upload_port = "rfm69gw.local"
upload_flags = --auth=Algernon1 --port 8266

ivankravets, 2018-02-05:
Fixed in https://github.com/platformio/platformio-core/commit/1362630ed64d7812b69609ac176788a8a11de340. 

Please open PlatformIO IDE Terminal and type `pio upgrade --dev`. Try to build a project again.

closed 2018-02-05 by ivankravets