platformio/platformio-atom-ide #1405: C/C++ project index rebuild failed

ramiws, 2018-01-19:
Cloning into '/Users/rami/.platformio/platforms/_tmp_installing-q5E12T-package'...
Submodule 'components/aws_iot/aws-iot-device-sdk-embedded-C' (https://github.com/espressif/aws-iot-device-sdk-embedded-C.git) registered for path 'components/aws_iot/aws-iot-device-sdk-embedded-C'
Submodule 'components/bt/lib' (https://github.com/espressif/esp32-bt-lib.git) registered for path 'components/bt/lib'
Submodule 'components/coap/libcoap' (https://github.com/obgm/libcoap.git) registered for path 'components/coap/libcoap'
Submodule 'components/esp32/lib' (https://github.com/espressif/esp32-wifi-lib.git) registered for path 'components/esp32/lib'
Submodule 'components/esptool_py/esptool' (https://github.com/espressif/esptool.git) registered for path 'components/esptool_py/esptool'
Submodule 'components/libsodium/libsodium' (https://github.com/jedisct1/libsodium.git) registered for path 'components/libsodium/libsodium'
Submodule 'components/micro-ecc/micro-ecc' (https://github.com/kmackay/micro-ecc.git) registered for path 'components/micro-ecc/micro-ecc'
Submodule 'components/nghttp/nghttp2' (https://github.com/nghttp2/nghttp2.git) registered for path 'components/nghttp/nghttp2'
Submodule 'components/spiffs/spiffs' (https://github.com/pellepl/spiffs.git) registered for path 'components/spiffs/spiffs'
Cloning into '/Users/rami/.platformio/platforms/_tmp_installing-q5E12T-package/components/aws_iot/aws-iot-device-sdk-embedded-C'...
Cloning into '/Users/rami/.platformio/platforms/_tmp_installing-q5E12T-package/components/bt/lib'...
Cloning into '/Users/rami/.platformio/platforms/_tmp_installing-q5E12T-package/components/coap/libcoap'...
Cloning into '/Users/rami/.platformio/platforms/_tmp_installing-q5E12T-package/components/esp32/lib'...
Cloning into '/Users/rami/.platformio/platforms/_tmp_installing-q5E12T-package/components/esptool_py/esptool'...
Cloning into '/Users/rami/.platformio/platforms/_tmp_installing-q5E12T-package/components/libsodium/libsodium'...
Cloning into '/Users/rami/.platformio/platforms/_tmp_installing-q5E12T-package/components/micro-ecc/micro-ecc'...
Cloning into '/Users/rami/.platformio/platforms/_tmp_installing-q5E12T-package/components/nghttp/nghttp2'...
Cloning into '/Users/rami/.platformio/platforms/_tmp_installing-q5E12T-package/components/spiffs/spiffs'...
Submodule 'third-party/mruby' (https://github.com/mruby/mruby) registered for path 'components/nghttp/nghttp2/third-party/mruby'
Submodule 'third-party/neverbleed' (https://github.com/h2o/neverbleed.git) registered for path 'components/nghttp/nghttp2/third-party/neverbleed'
Cloning into '/Users/rami/.platformio/platforms/_tmp_installing-q5E12T-package/components/nghttp/nghttp2/third-party/mruby'...
Cloning into '/Users/rami/.platformio/platforms/_tmp_installing-q5E12T-package/components/nghttp/nghttp2/third-party/neverbleed'...
Error: Traceback (most recent call last):
  File "/Users/rami/.platformio/penv/lib/python2.7/site-packages/platformio/__main__.py", line 120, in main
    cli(None, None, None)
  File "/Users/rami/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/Users/rami/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/Users/rami/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/rami/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/rami/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/Users/rami/.platformio/penv/lib/python2.7/site-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Users/rami/.platformio/penv/lib/python2.7/site-packages/platformio/commands/init.py", line 102, in cli
    pg = ProjectGenerator(project_dir, ide, env_name)
  File "/Users/rami/.platformio/penv/lib/python2.7/site-packages/platformio/ide/projectgenerator.py", line 38, in __init__
    self._gather_tplvars()
  File "/Users/rami/.platformio/penv/lib/python2.7/site-packages/platformio/ide/projectgenerator.py", line 155, in _gather_tplvars
    self._tplvars.update(self.get_project_build_data())
  File "/Users/rami/.platformio/penv/lib/python2.7/site-packages/platformio/util.py", line 134, in __call__
    value = self.func(*args)
  File "/Users/rami/.platformio/penv/lib/python2.7/site-packages/platformio/ide/projectgenerator.py", line 79, in get_project_build_data
    raise result.exception
IOError: [Errno 2] No such file or directory: '/Users/rami/.platformio/platforms/esp-idf/platform.json'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

ivankravets, 2018-01-30:
What is your `platformio.ini`?

ivankravets, 2018-06-20:
Please reopen if you still need help

closed 2018-06-20 by ivankravets

DoktorDaveJoos, 2018-09-17:
Had the same *error* message.

Steps to reproduce:

1. `mkdir iot-project && cd iot-project`
2. `virtualenv .`
3. `mkdir iot-device && cd iot-device`
4. `pip install -U platformio`
5. `pio platform install https://github.com/esp8266/Arduino.git`

It worked after i changed the `platformio.ini` to
```
[env:nodemcuv2]
platform = https://github.com/platformio/platform-espressif8266.git#feature/stage
board = nodemcuv2
framework = arduino
```
-> i could simply run `pio run` and it worked out.

But the *error* message was quite confusing.. Hope this helps.

ivankravets, 2018-09-17:
> pio platform install https://github.com/esp8266/Arduino.git

This is not a platform, this is a framework.