platformio/platformio-atom-ide #1417: C/C++ project index rebuild failed

baldwin1974, 2018-01-21:
Error: [Sun Jan 21 09:48:47 2018] Processing disco_f469ni (platform: ststm32; board: disco_f469ni; framework: mbed)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Collected 16 compatible libraries
Scanning dependencies...
Error: Could not find `ualloc` dependency for `mbed-drivers` library
========================== [ERROR] Took 3.79 seconds ==========================

NOTES: (1) can't view PIO Core Call error log (no scrolling possible)
(2) this occurs on initiating a new project

ivankravets, 2018-01-30:
Have you resolved this issue?

baldwin1974, 2018-01-30:
I could not get platformio to recognize the library, so I took a break.  I think, however, it must be user error, as I am pretty new at this.  I will close the issue.

closed 2018-01-30 by baldwin1974

ivankravets, 2018-04-25:
Duplicate of https://github.com/platformio/platformio-atom-ide/issues/1356