platformio/platformio-atom-ide #1442: C/C++ project index rebuild failed

n1x0id, 2018-01-27:
Error: [Sat Jan 27 13:07:31 2018] Processing uno (platform: atmelavr; lib_deps: Adafruit Unified Sensor, DHT sensor library, DallasTemperature, OneWire, PubSubClient, Ethernet, ArduinoLog, ArduinoJson; board: uno; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
System: ATMEGA328P 16MHz RAM/2K Flash/31.50K
UnicodeEncodeError: 'ascii' codec can't encode characters in position 410-419: ordinal not in range(128):
File "/Users/nixoid/.platformio/penv/lib/python2.7/site-packages/platformio/builder/main.py", line 158:
env.SConscript("$BUILD_SCRIPT")
File "/Users/nixoid/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 551:
return _SConscript(self.fs, *files, **subst_kw)
File "/Users/nixoid/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 256:
call_stack[-1].globals)
File "/Users/nixoid/.platformio/platforms/atmelavr/builder/main.py", line 224:
target_elf = env.BuildProgram()
File "/Users/nixoid/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/Users/nixoid/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/platformio.py", line 63:
env.BuildFrameworks(env.get("PIOFRAMEWORK"))
File "/Users/nixoid/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/Users/nixoid/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/platformio.py", line 261:
env.ConvertInoToCpp()
File "/Users/nixoid/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/Users/nixoid/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/piomisc.py", line 189:
out_file = c.convert(ino_nodes)
File "/Users/nixoid/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/piomisc.py", line 54:
return self.process(contents)
File "/Users/nixoid/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/piomisc.py", line 77:
assert self._gcc_preprocess(contents, out_file)
File "/Users/nixoid/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/piomisc.py", line 88:
fp.write(contents)
========================== [ERROR] Took 1.27 seconds ==========================

n1x0id, 2018-01-27:
.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/piomisc.py:88
fp.write(contents) -> fp.write(contents.encode('utf-8'))

ivankravets, 2018-01-27:
Duplicate in https://github.com/platformio/platformio-atom-ide/issues/1438

ivankravets, 2018-01-27:
Thanks that reported!

closed 2018-01-27 by ivankravets