platformio/platformio-atom-ide #1444: C/C++ project index rebuild failed

YannikW, 2018-01-27:
Im getting following error on an existing project which worked fine a couple days ago..
I tried reinstalling platformio or using VS code instead of atom, but everytime the same error..

Is looks like theres a bug in some Platformio Phyton files, isn't it?

What can I do to compile my code?

```
Error: [01/27/18 11:47:09] Processing esp12e (platform: espressif8266; board_f_flash: 80000000L; board_f_cpu: 160000000L; board: esp12e; framework: arduino)

Verbose mode can be enabled via `-v, --verbose` option
System: ESP8266 160MHz RAM/80K Flash/4M
UnicodeEncodeError: 'ascii' codec can't encode character u'\xb4' in position 28148: ordinal not in range(128):
File "C:\users\yannik\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 158:
env.SConscript("$BUILD_SCRIPT")
File "C:\Users\Yannik\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 551:
return _SConscript(self.fs, *files, **subst_kw)
File "C:\Users\Yannik\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 256:
call_stack[-1].globals)
File "C:\users\yannik\.platformio\platforms\espressif8266\builder\main.py", line 353:
target_elf = env.BuildProgram()
File "C:\Users\Yannik\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\users\yannik\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 63:
env.BuildFrameworks(env.get("PIOFRAMEWORK"))
File "C:\Users\Yannik\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\users\yannik\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 261:
env.ConvertInoToCpp()
File "C:\Users\Yannik\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "C:\users\yannik\.platformio\penv\lib\site-packages\platformio\builder\tools\piomisc.py", line 189:
out_file = c.convert(ino_nodes)
File "C:\users\yannik\.platformio\penv\lib\site-packages\platformio\builder\tools\piomisc.py", line 54:
return self.process(contents)
File "C:\users\yannik\.platformio\penv\lib\site-packages\platformio\builder\tools\piomisc.py", line 77:
assert self._gcc_preprocess(contents, out_file)
File "C:\users\yannik\.platformio\penv\lib\site-packages\platformio\builder\tools\piomisc.py", line 88:
fp.write(contents)
 [ERROR] Took 0.82 seconds 
```

YannikW, 2018-01-27:
Workaround in https://github.com/platformio/platformio-atom-ide/issues/1442 helped for me

ivankravets, 2018-01-27:
Duplicate in https://github.com/platformio/platformio-atom-ide/issues/1438

closed 2018-01-27 by ivankravets